react-native-camera/react-native-camera #1963: Onfaces detected is getting called when faces comes infront of camera but doesnt fire event when the faces go out of camera?

shohailahm, 2018-12-02:
# Bug Report
**To Do First**
- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Related Modules**

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

n1ru4l, 2018-12-02:
Please fill out the template.

shohailahm, 2018-12-10:
someone found anything guys?

closed 2019-06-08 by stale[bot]