react-native-camera/react-native-camera #1964: Uploading video to server

jamesawer3, 2018-12-03:
Hi,
 I'm using react-native-camera(^0.6.0) in my app, I have to upload the recorded video to server as multipart data. In android they are posting like this,

```
{
  _parts: [
    [
      'file',
      {
        name: 'VID_20181130_150959.mp4',
        uri: 'file:///storage/emulated/0/DCIM/VID_20181130_150959.mp4',
        type: 'video/mp4'
      }
    ]
  ]
}
```

but in iOS the file path is returning as `assets-library://asset/asset.mov?id=41B76A24-1018-46C1-A658-C1EFFC552FD0&ext=mov` but if I post the assets path it's not uploading.

PostData:

```
{
  _parts: [
    [
      'file',
      {
        name: '41B76A24-1018-46C1-A658-C1EFFC552FD0.mov',
        uri: 'assets-library://asset/asset.mov?id=41B76A24-1018-46C1-A658-C1EFFC552FD0.mov',
        type: 'video/mov'
      }
    ]
  ]
}
```

Can anyone help me out with this????

n1ru4l, 2018-12-12:
You will have to provide more context. Otherwise we won't be able to help you

mudasserhassan1, 2018-12-24:
Android needs Absolute file path (which you are providing) while IOS needs 'uri' of that image/video to pass to API in my case it was 'content://media/external/video/media/120529' .
Hope this helps.

n1ru4l, 2018-12-26:
Going to close this since there is no response since 24 days.

closed 2018-12-26 by n1ru4l