react-native-camera/react-native-camera #1965: How to know if camera is started recording

svipben, 2018-12-03:
There's no such var like `isRecording` or something like that which returns boolean if the recording is started or no.

sibelius, 2018-12-03:
can't you handle this on javascript side?

svipben, 2018-12-03:
I'm doing it but my problem is I get `Video is not recording.` by trying to use `stopRecording()`. This happens if I click too often on start/stop recording. Even if I use var by checking if the recording is in progress I get the same issue. Since I can't be sure 100% if it's started because I'm setting it only before `recordAsync`, I believe boolean from the native module will be way better.

sibelius, 2018-12-03:
can you avoid using state in this case

you can have `this.isRecording` that will be faster than `setState`, so you won't have any problem I guess

svipben, 2018-12-04:
Okay, the issue is when I start recording there's some sort of glitch seems like it initializes the camera and if I press stop that moment I get `Video is not recording.`, but second time everything seems okay.

svipben, 2018-12-04:
I'm currently working on iOS implementation and seems everything is good, I can't test the Android because currently I don't have it, but I can make a PR for iOS if needed.

svipben, 2018-12-04:
Sounds good!

svipben, 2018-12-04:
Already opened a PR here: https://github.com/react-native-community/react-native-camera/pull/1969

n1ru4l, 2018-12-05:
:tada: This issue has been resolved in version 1.5.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.5.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

closed 2018-12-05 by n1ru4l

nachumFreedman, 2019-09-22:
Hey guys can we do our best to make vitally useful APIs work for both IOS and Android...

n1ru4l, 2019-09-23:
@nachumFreedman Yes definitely! Please send a pull request. Please also notice that this project is based upon user contributions. Do not assume just opening an issue will solve the problem (in the near future).

I am going to lock this issue. In case there is anything further to discuss, please open a new issue or even better: Let's continue the discussion in a pull request that brings this feature to the android platform.