react-native-camera/react-native-camera #1967: taking picture with takePictureAsync immediately after changing camera type doesn't work

Luke-Tan, 2018-12-04:
I am trying to take a photo using react native camera immediately after changing from back to front camera. To give you an idea, the code looks kind of like this:

```
state = {
    type: Camera.Constants.Type.back
}

takePicture = () => {
    const options = {quality:0.5}
    this.setState({type:Camera.Constants.Type.front}, () => {
        this.camera.takePictureAsync(options).then(data => {
            console.log(data)
        });
    })
};
```
I presumed that the photo I would have taken would be the one with the front facing camera. However, the photo i get is always the one with the back facing camera. Is there something im missing?

n1ru4l, 2018-12-12:
Please use the issue template.

There is a certain delay because the communication between the js thread and native thread is asynchronous. In the future this might be solvable with react-native fabric (read more here: https://github.com/react-native-community/discussions-and-proposals/issues/4 keyword JSI Bindings).

Luke-Tan, 2018-12-12:
So is the only way to fix this to add a setTimeout to takePicture after calling setState?

closed 2019-06-08 by stale[bot]