react-native-camera/react-native-camera #1970: iOS: Photo orientation issue

Orxcle-xx, 2018-12-04:
# Bug Report

- [ ] Did you try latest release?, Yes: 1.4.3
- [ ] Did you try master? Yes
- [ ] Did you look for existing matching issues? Yes, lots of issues found, maybe some of them are related to mine. No solution found for me.

**Related Modules**

Using only this module.

**Platforms**

Haven't tried on Android. Working currently only with iOS.

**Versions**
- iOS: iOS 12
- react-native-camera: 1.4.3
- react-native: 0.57.5
- react: 16.6.1
 -react-navigation: 3.0.0

**Description/Current Behaviour**
I make a photo using camera. When I display this photo using a library gl-react / gl-react-native.
My photo is rotated. I tried to use forceUpOrientation -- _photo changes the rotation_ :) but it's still rotated.
I tried to put an image ( inside Surface container from gl-react-native) using uri and using base64.
No luck. I tried even to put it via GLImage from gl-react-image. No luck.
By the way, if I use standard <Image/> tag -- photo is displayed correctly.

**Expected Behaviour**
Photo is not rotated and displayed as is.

**Steps to Reproduce**
Well, see Description.

**Additionals**
Thank you for the great library.

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

n1ru4l, 2018-12-12:
Have you tried using the `orientation` option?

sibelius, 2018-12-14:
recheck on 1.6.3

closed 2018-12-14 by sibelius