react-native-camera/react-native-camera #1975: Cannot focus on Samsung Galaxy J1 2016

jacktv1, 2018-12-06:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
I have test camera on Samsung galaxy J1 2016 and camera can not focus, i can not scan bardcode but other is work expectly.
Have any one have the same issue, can help me about this problem.

My enviroment:
react-native: 0.57.7
react-native-camera: 1.5.0

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

mmamaniwolox, 2019-02-09:
Im having this issue as well on a J1 and a Google Pixel

agrass, 2019-05-23:
Same problem with Galaxy Tab A (SM-T280) with RN-Camera 2.0.1. Any idea on how to fix this?

agrass, 2019-05-31:
Fixed updating to last version and using the touch screen focus.

closed 2019-08-07 by stale[bot]