react-native-camera/react-native-camera #1977: `NativeModules.CameraManager` is undefined in swift projects(but not obj-c projects)

augustl, 2018-12-06:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

RNCamera

**Platforms**

iOS

**Versions**

- Android: -
- iOS: 11.4
- react-native-camera: 1.5.0
- react-native: 0.57.7
- react: 16.6.3

**Description/Current Behaviour**

When integrating react-native-camera into an existing iOS app, it seems to work fine in a plain objective-c project, but not in a plain swift project.

When actually attempting to use react-native-camera, an error is raised because it can't find NativeModules.CameraManager or NativeModules.CameraModule.

I'm using cocoapods to set up the Xcode projects with react-native as well as react-native-camera.

The actual error you end up getting is:

`undefined is not an object (evaluating 'CameraManager.Aspect')`

This is because `src/Camera.js` defines CameraManager as `const CameraManager = NativeModules.CameraManager || NativeModules.CameraModule;`, both of which are undefined in a swift project.

**Expected Behaviour**

`NativeModules.CameraManager`  should be available in both obj-c projects and swift projects.

**Steps to Reproduce**

I created this repo to demonstrate the issue. It should be quite easy to reproduce, by following the steps in the README.

https://github.com/augustl/react-native-camera-ios-native-module-issue

The commit log should show pretty clearly what changes I've done. For example, [here's the commit that adds react-native](https://github.com/augustl/react-native-camera-ios-native-module-issue/commit/c39f1dba31141ee1b880168a6356476ced96b703) to the obj-c part of the project, and [here's the one that adds react-native-camera](https://github.com/augustl/react-native-camera-ios-native-module-issue/commit/824170fdc924312e89f184fa8a6c0fb08d6f89cf).

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

augustl, 2018-12-06:
FWIW - this is a brand new app, so I decided to roll it as an objective-c app instead of a swift app. I can very easily create swift classes in my objective-c based app, so as far as I can tell there are no downsides to doing that :)

saadrehman013, 2018-12-27:
Any update on this issue. Actually, I am using 'react-native-beautiful-video-recorder' and this module resides in the peer dependencies. I am stuck with this error. 

undefined is not an object (evaluating 'CameraManager.Aspect')

Any help??

cydjohn, 2019-01-07:
I have the same issue, solved by running `pod install` after `react-native link react-native-camera`.

sibelius, 2019-01-08:
Can we have an example-swift in this repo to make it easy to reproduce this bug?

augustl, 2019-01-08:
My repo has examples for both swift and obj-c :) https://github.com/augustl/react-native-camera-ios-native-module-issue

n1ru4l, 2019-01-08:
@sibelius Sure but adding something that is not working will not help 😅

AFAIK there is no official react-native swift template/create-react-native-app template

Ramon-Lobo, 2019-03-01:
@augustl I had the same problem and solved it by adding `RNCamera` to my Xcode Project's libraries folder. It is located in `node_modules/react-native-camera/ios`.

Hope it helps you!

rainbowql, 2019-03-23:
any solutions?  I had the same problem....

icxcat, 2019-04-22:
i solved it by Manual install step by step

closed 2019-06-28 by stale[bot]