react-native-camera/react-native-camera #1978: Setting RNCamera captureAudio to false does no not work.

whydna, 2018-12-06:
# Bug Report
**To Do First**
- [Y ] Did you try latest release?
- [Y ] Did you try master?
- [Y ] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
iOS

**Versions**
- iOS:
- react-native-camera: 1.5.0

**Description/Current Behaviour**
Setting `captureAudio` to `false` doesn't work. It still requests microphone permissions and the output video still contains audio track.

```
<RNCamera
                    ref={ref => {
                        this.camera = ref
                    }}
                    style={styles.preview}
                    type={RNCamera.Constants.Type.back}
                    captureAudio={false}
                />
```

**Expected Behaviour**
If `captureAudio` is set to `false`, the camera should not request microphone permissions, and the output video should not contain an audio track.

**Steps to Reproduce**
On any iPhone X or Xs or XsMax device:

```
<RNCamera
                    ref={ref => {
                        this.camera = ref
                    }}
                    style={styles.preview}
                    type={RNCamera.Constants.Type.back}
                    captureAudio={false}
                />
```

Output video contains audio.

n1ru4l, 2018-12-11:
The `captureAudio` property is not implemented for `RNCamera`. I updated the documentation. You would want to use the `mute` property.

closed 2018-12-11 by n1ru4l

n1ru4l, 2018-12-12:
:tada: This issue has been resolved in version 1.6.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.6.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

n1ru4l, 2018-12-18:
**Update**: The `captureAudio` option is used to check whether `RNCamera` should ask for permissions Android Permissions (https://github.com/react-native-community/react-native-camera/blob/e79cd566ab61e31105dddef0193d7dd351a14d59/src/RNCamera.js#L412) or not.

gwl002, 2019-05-16:
I set captureAudio prop to false,but on ios It still ask for microphone permission,and on android throw yellow warning "Otherwise you should set captureAudio to false".But I have set it to false already! @n1ru4l

n1ru4l, 2019-05-16:
@gwl002 I am currently not maintaining react-native-camera. The issue and a possible way to resolve it is documented in https://github.com/react-native-community/react-native-camera/issues/2085

If you are using this library I am encouraging you to resolve this issue, I think other people here would definitely appreciate your work.

For now I am going to lock this conversation.

PR #1987 (2018-12-11): docs(rn-camera): remove unimplemented property
Fixes #1978
commit 515bccb7 (2018-12-11): docs(rn-camera): remove unimplemented property (#1987) [skip ci]

Fixes #1978