react-native-camera/react-native-camera #1980: add react-native-camera-cli that makes common tasks easier

n1ru4l, 2018-12-07:
# Feature-Request

**Describe the Feature**
Add a react-native-camera-cli that can automate certain install steps.

**Possible Implementations**
First command:

- `react-native-camera download-ios-vision-frameworks`

A script that downloads all necessary `*.frameworks` and adds them to the `*.pbxproj`.
This could e.g. be executed as a `postInstall` script.

**Related Issues**
<Insert all issues about people having trouble with setting up react-native-camera and face recognition here>

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2018-12-07:
check this https://github.com/facebook/react-native-fbsdk/blob/master/bin/ios_setup.js

n1ru4l, 2018-12-07:
Dope thank you!

closed 2019-06-08 by stale[bot]