react-native-camera/react-native-camera #1982: Example RN camera is crashing on IOS and android

radjivF, 2018-12-10:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
Android
iOS

**Versions**

- Android: last one 
- iOS: last one
- react-native-camera: 1.5.1
- react-native:0.55.4
- react:16.3.1

**Description/Current Behaviour**

Im using the tutorial for RNCamera and when I render its just crashes without error message

**Expected Behaviour**
Not crash
**Steps to Reproduce**
Just using the example on this repo

Please help

n1ru4l, 2018-12-10:
Do you mean https://github.com/react-native-community/rncamera-example ?

jetonk, 2018-12-10:
I have the same problem

n1ru4l, 2018-12-10:
I am currently working on porting the rncamera-example to this repository, it seems like the face detection stuff is broken.

n1ru4l, 2018-12-12:
@radjivF Have you tried the example app that is in this repository?

radjivF, 2018-12-12:
@n1ru4l yes I have tried the example, it works on ios but still crash on android.

n1ru4l, 2018-12-13:
I have no problems with https://github.com/react-native-community/react-native-camera/tree/master/examples/basic

Can you please provide more information about your hardware?

emilemoureau, 2018-12-13:
I have the same problem..

n1ru4l, 2018-12-13:
@emilemoureau Crash logs?

closed 2018-12-13 by sibelius

emilemoureau, 2018-12-13:
Solved, sorry, I didn't have the right config in info.plist !

muhammadridho, 2019-04-25:
i have same problem. working device Xiaomi Mi5. when i first running i get crash screen

[image]

mMarcos208, 2019-12-13:
I have the same problem in IOS `13.3`