react-native-camera/react-native-camera #1988: [Android] recordAsync Promise never resolves if device is rotated

n1ru4l, 2018-12-11:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

RNCamera

**Platforms**

Android

**Versions**

- Android: 7.1.1
- iOS: - 
- react-native-camera: master
- react-native: 0.57.1
- react: 16.5.0

**Steps to Reproduce**

1. Start the example app with no `screenOrientation` defined in the `AndroidManifest.xml` and no locked portrait mode on the phone

2. Press the `REC` button and immediately afterwards (or any time before the recording stops) rotate the phone so that view also rotates

**Description/Current Behaviour**

The promise returned by `recordAsync` does never resolve.

**Expected Behaviour**

The promise returned by `recordAsync` does resolve after the recording has finished.

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2018-12-14:
@n1ru4l is this still a thing?

n1ru4l, 2018-12-14:
Yes had no time to dig deeper as it is not a requirement of my app (locked in portrait)

punksta, 2019-01-10:
When device is rotated without `screenOrientation` `android:configChanges` in manifest android os [recreates](https://developer.android.com/guide/topics/resources/runtime-changes) activity. it means the whole react-native ui will be recreated. I am not 100% sure, but I it seams like js engine is getting restarted in this case and we can't change it. 
So I think opposite, all pending promises should be rejected here: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraView.java#L495

n1ru4l, 2019-01-11:
@punksta can you contribute a fix for this issue?

punksta, 2019-01-11:
@n1ru4l btw, why you need no screenOrientation in manifest? https://stackoverflow.com/a/5726776/2795206 you can have callbacks for orientation changes.

n1ru4l, 2019-01-11:
@punksta I was testing how I could reproduce this issue. My product does not suffer from this issue, but other projects might do so.

closed 2019-06-08 by stale[bot]