react-native-camera/react-native-camera #1999: FAILURE: Build failed with an exception.

Ann1995, 2018-12-14:
* What went wrong:
Could not resolve all files for configuration ':react-native-camera:debugCompileClasspath'.
> Could not find play-services-vision.jar (com.google.android.gms:play-services-vision:15.0.2).
  Searched in the following locations:
      https://jcenter.bintray.com/com/google/android/gms/play-services-vision/15.0.2/play-services-vision-15.0.2.jar

n1ru4l, 2018-12-14:
@Ann1995 Did you follow the instructions in the `Readme`? https://github.com/react-native-community/react-native-camera#android

closed 2018-12-14 by sibelius

Ann1995, 2018-12-14:
@n1ru4l  yes.. my package.json 
{
  "name": "spector",
  "version": "1.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "test": "jest",
    "lint": "eslint app --fix",
    "postinstall": "./fix-rncamera.sh ./node_modules/react-native-camera/ios/RNCamera.xcodeproj/project.pbxproj"
  },
  "rnpm": {
    "assets": [
      "./app/assets/fonts"
    ]
  },
  "dependencies": {
    "@babel/core": "7.0.0-beta.47",
    "@babel/plugin-proposal-decorators": "^7.0.0-beta.47",
    "@babel/plugin-transform-runtime": "7.0.0-beta.47",
    "@babel/register": "7.0.0-beta.47",
    "@babel/runtime": "^7.1.5",
    "axios": "^0.18.0",
    "babel-plugin-transform-decorators-legacy": "^1.3.5",
    "enzyme-to-json": "^3.3.5",
    "jest-serializer-enzyme": "^1.0.0",
    "metro-react-native-babel-preset": "^0.49.2",
    "moment": "^2.22.2",
    "prop-types": "^15.6.1",
    "react": "16.6.0-alpha.8af6728",
    "react-addons-test-utils": "^15.6.2",
    "react-native": "^0.57.7",
    "react-native-blur": "^3.2.2",
    "react-native-camera": "^1.2.0",
    "react-native-config": "^0.11.5",
    "react-native-image-crop-picker": "^0.20.3",
    "react-native-keyboard-aware-scroll-view": "^0.7.2",
    "react-native-localization": "^1.0.7",
    "react-native-navigation": "^1.1.446",
    "react-native-photo-view": "^1.5.2",
    "react-native-qrcode-scanner": "^1.0.1",
    "react-native-router-flux": "^4.0.1",
    "react-native-vector-icons": "6.0.2",
    "react-redux": "^5.0.7",
    "recompose": "^0.27.0",
    "redux": "^4.0.0",
    "redux-devtools-extension": "^2.13.5",
    "redux-persist": "^5.9.1",
    "redux-persist-filesystem-storage": "^1.3.2",
    "redux-thunk": "^2.2.0",
    "reselect": "^3.0.1",
    "rn-fetch-blob": "^0.10.13"
  },
  "devDependencies": {
    "@babel/core": "^7.0.0-beta.47",
    "babel-core": "^7.0.0-bridge.0",
    "babel-eslint": "^8.2.3",
    "babel-jest": "23.4.0",
    "babel-loader": "^8.0.4",
    "enzyme": "^3.3.0",
    "enzyme-adapter-react-16": "^1.1.1",
    "eslint": "^4.19.1",
    "eslint-config-airbnb": "^16.1.0",
    "eslint-plugin-import": "^2.11.0",
    "eslint-plugin-jsx-a11y": "^6.0.3",
    "eslint-plugin-react": "^7.7.0",
    "jest": "23.4.0",
    "moxios": "^0.4.0",
    "react-dom": "^16.6.3",
    "react-native-mock-render": "^0.1.2",
    "react-test-renderer": "^16.4.0",
    "redux-mock-store": "^1.5.3"
  },
  "jest": {
    "preset": "react-native",
    "setupTestFrameworkScriptFile": "<rootDir>/testsconfig.js",
    "snapshotSerializers": [
      "./node_modules/enzyme-to-json/serializer"
    ],
    "transformIgnorePatterns": [
      "node_modules/(?!(jest-)?react-native|react-navigation|rn-fetch-blob)"
    ],
    "collectCoverage": true,
    "coverageReporters": [
      "lcov"
    ]
  }
}

n1ru4l, 2018-12-14:
I ran the following steps to ensure everything is alright:

```bash
# ensure gradle caches are busted
rm -rf $HOME/.gradle/caches/
# clone repository
git clone https://github.com/react-native-community/react-native-camera.git
cd react-native-camera
# insteall react-native-camera dependencies
yarn
# go into example
cd examples/basic
# install sample dependencies
yarn
# run android build
yarn react-native run-android
```

<details>

<summary>Full Build Output</summary>

```
$ /Users/laurin/projects/react-native-camera/examples/basic/node_modules/.bin/react-native run-android
JS server already running.
Building and installing the app on the device (cd android && ./gradlew installDebug)...
Starting a Gradle Daemon (subsequent builds will be faster)
Download https://maven.google.com/com/android/tools/build/gradle/3.0.1/gradle-3.0.1.pom
Download https://maven.google.com/com/android/tools/build/gradle-core/3.0.1/gradle-core-3.0.1.pom
Download https://jcenter.bintray.com/com/android/tools/build/transform-api/2.0.0-deprecated-use-gradle-api/transform-api-2.0.0-deprecated-use-gradle-api.pom
Download https://jcenter.bintray.com/org/jetbrains/kotlin/kotlin-stdlib/1.1.3-2/kotlin-stdlib-1.1.3-2.pom
Download https://jcenter.bintray.com/org/ow2/asm/asm-analysis/5.1/asm-analysis-5.1.pom
Download https://jcenter.bintray.com/org/ow2/asm/asm-commons/5.1/asm-commons-5.1.pom
Download https://jcenter.bintray.com/org/ow2/asm/asm-parent/5.1/asm-parent-5.1.pom
Download https://jcenter.bintray.com/org/ow2/ow2/1.3/ow2-1.3.pom
Download https://jcenter.bintray.com/org/jacoco/org.jacoco.core/0.7.4.201502262128/org.jacoco.core-0.7.4.201502262128.pom
Download https://jcenter.bintray.com/org/ow2/asm/asm-util/5.1/asm-util-5.1.pom
Download https://jcenter.bintray.com/org/jacoco/org.jacoco.build/0.7.4.201502262128/org.jacoco.build-0.7.4.201502262128.pom
Download https://jcenter.bintray.com/org/ow2/asm/asm/5.1/asm-5.1.pom
Download https://jcenter.bintray.com/org/jacoco/org.jacoco.report/0.7.4.201502262128/org.jacoco.report-0.7.4.201502262128.pom
Download https://jcenter.bintray.com/net/sf/jopt-simple/jopt-simple/4.9/jopt-simple-4.9.pom
Download https://jcenter.bintray.com/com/google/protobuf/protobuf-java/3.0.0/protobuf-java-3.0.0.pom
Download https://jcenter.bintray.com/org/sonatype/oss/oss-parent/7/oss-parent-7.pom
Download https://jcenter.bintray.com/net/sf/proguard/proguard-gradle/5.3.3/proguard-gradle-5.3.3.pom
Download https://jcenter.bintray.com/com/google/protobuf/protobuf-parent/3.0.0/protobuf-parent-3.0.0.pom
Download https://jcenter.bintray.com/net/sf/proguard/proguard-parent/5.3.3/proguard-parent-5.3.3.pom
Download https://jcenter.bintray.com/com/google/google/1/google-1.pom
Download https://maven.google.com/com/android/databinding/compilerCommon/3.0.1/compilerCommon-3.0.1.pom
Download https://maven.google.com/com/android/tools/build/gradle-api/3.0.1/gradle-api-3.0.1.pom
Download https://maven.google.com/com/android/tools/external/org-jetbrains/uast/26.0.1/uast-26.0.1.pom
Download https://maven.google.com/com/android/tools/lint/lint/26.0.1/lint-26.0.1.pom
Download https://maven.google.com/com/android/tools/build/builder/3.0.1/builder-3.0.1.pom
Download https://maven.google.com/com/android/tools/analytics-library/protos/26.0.1/protos-26.0.1.pom
Download https://maven.google.com/com/android/tools/build/builder-test-api/3.0.1/builder-test-api-3.0.1.pom
Download https://maven.google.com/com/android/tools/build/manifest-merger/26.0.1/manifest-merger-26.0.1.pom
Download https://maven.google.com/com/android/tools/sdk-common/26.0.1/sdk-common-26.0.1.pom
Download https://maven.google.com/com/android/tools/sdklib/26.0.1/sdklib-26.0.1.pom
Download https://jcenter.bintray.com/com/squareup/javawriter/2.5.0/javawriter-2.5.0.pom
Download https://maven.google.com/com/android/tools/common/26.0.1/common-26.0.1.pom
Download https://maven.google.com/com/android/tools/ddms/ddmlib/26.0.1/ddmlib-26.0.1.pom
Download https://jcenter.bintray.com/org/bouncycastle/bcpkix-jdk15on/1.56/bcpkix-jdk15on-1.56.pom
Download https://maven.google.com/com/android/tools/build/builder-model/3.0.1/builder-model-3.0.1.pom
Download https://jcenter.bintray.com/org/bouncycastle/bcprov-jdk15on/1.56/bcprov-jdk15on-1.56.pom
Download https://jcenter.bintray.com/com/googlecode/json-simple/json-simple/1.1/json-simple-1.1.pom
Download https://jcenter.bintray.com/it/unimi/dsi/fastutil/7.2.0/fastutil-7.2.0.pom
Download https://jcenter.bintray.com/org/ow2/asm/asm-tree/5.1/asm-tree-5.1.pom
Download https://maven.google.com/com/android/tools/analytics-library/tracker/26.0.1/tracker-26.0.1.pom
Download https://maven.google.com/com/android/tools/analytics-library/shared/26.0.1/shared-26.0.1.pom
Download https://maven.google.com/com/android/tools/build/apksig/3.0.1/apksig-3.0.1.pom
Download https://jcenter.bintray.com/org/eclipse/jdt/core/compiler/ecj/4.6.1/ecj-4.6.1.pom
[... 239 lines cut ...]
Download https://maven.google.com/com/android/support/support-fragment/27.1.0/support-fragment-27.1.0.aar
Download https://maven.google.com/com/android/support/support-core-ui/27.1.0/support-core-ui-27.1.0.aar
Download https://maven.google.com/com/android/support/support-core-utils/27.1.0/support-core-utils-27.1.0.aar
Download https://maven.google.com/com/android/support/support-compat/27.1.0/support-compat-27.1.0.aar

> Task :react-native-camera:compileDebugJavaWithJavac
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
[... cut ...]
Note: Recompile with -Xlint:unchecked for details.

> Task :app:installDebug
Installing APK 'app-debug.apk' on 'Nexus 5X - 7.1.1' for app:debug
Installed on 1 device.

BUILD SUCCESSFUL in 2m 7s
```

</details>

___

Seems like you misconfigured something. Please double check your `android/build.gradle` and `android/app/build.gradle` with the ones of the example.

It is most likely you forgot to add a repository since the google vision library cannot be found in the jcenter registry (because it does not exist there).

Ann1995, 2018-12-14:
android/build.gradle
// Top-level build file where you can add configuration options common to all sub-projects/modules.

```
buildscript {
    repositories {
        jcenter()
        google()
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.2.1'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

 allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
        maven { url "https://jitpack.io" }
    }
} 

ext {
    buildToolsVersion = "28.0.3"
    minSdkVersion = 16
    compileSdkVersion = 27
    targetSdkVersion = 27
    supportLibVersion = "26.1.0"
}

subprojects {
    project.evaluationDependsOn(':app')
    project.configurations.all {
        resolutionStrategy.eachDependency { details ->
           if (details.requested.group == 'com.android.support'&& !details.requested.name.contains('multidex') ) 
           {
                details.useVersion "26.1.0"
            }
        }
    }
}  
```

android/app/build.gradle
apply plugin: "com.android.application"

import com.android.build.OutputFile

/**
 * The react.gradle file registers a task for each build variant (e.g. bundleDebugJsAndAssets
 * and bundleReleaseJsAndAssets).
 * These basically call `react-native bundle` with the correct arguments during the Android build
 * cycle. By default, bundleDebugJsAndAssets is skipped, as in debug/dev mode we prefer to load the
 * bundle directly from the development server. Below you can see all the possible configurations
 * and their defaults. If you decide to add a configuration block, make sure to add it before the
[... 136 lines cut ...]
    compile project(':react-native-navigation')
    compile project(':react-native-config')
    compile project(':react-native-localization')
    compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        compile 'com.android.support:exifinterface:25.+'
        compile ('com.google.android.gms:play-services-vision:12.0.1') {
            force = true
        }
    }
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:${rootProject.ext.supportLibVersion}"
    compile "com.facebook.react:react-native:0.57.7"  // From node_modules
 }

// Run this once to be able to run the application with BUCK
// puts all compile dependencies into folder libs for BUCK to use
task copyDownloadableDepsToLibs(type: Copy) {
    from configurations.compile
    into 'libs'
}

```

I tried to change    compile ('com.google.android.gms:play-services-vision:12.0.1') =>   compile ('com.google.android.gms:play-services-vision:15.0.2') but it did not help

n1ru4l, 2018-12-14:
@Ann1995 check https://github.com/react-native-community/react-native-camera/issues/1878 please

Ann1995, 2018-12-14:
@n1ru4l thanks for your help!!!

n1ru4l, 2018-12-14:
cc @sibelius We need to document this somehow. Why does this issue not occur with the example app?

dksami, 2019-07-29:
I have tried every single step and it seems not to be working. Can anyone help me please I
 am getting this error and after hours of search i am still unable to reach a solution

[image]