react-native-camera/react-native-camera #2000: Package is overweighted

Nabellaleen, 2018-12-14:
Hi,

After a npm install of your package, the directory size is 16M, mostly due to android build directory, embedding stuff which should not be commited

You should clean that, it's totally useless and horrible for users :)

```
du node_modules/react-native-camera/android/ -h -d 1
68K	node_modules/react-native-camera/android/gradle
4,2M	node_modules/react-native-camera/android/.gradle
596K	node_modules/react-native-camera/android/src
9,5M	node_modules/react-native-camera/android/build
15M	node_modules/react-native-camera/android/
```

sibelius, 2018-12-14:
android/build is generated by android

n1ru4l, 2018-12-14:
I think he is right run 

```
npm pack react-native-camera | grep android/build

```

sibelius, 2018-12-14:
we can ignore more stuff on .npmignore

like examples

sibelius, 2018-12-14:
https://stackoverflow.com/questions/40642008/how-do-i-view-the-size-of-npm-packages

> react-native-camera@1.6.3: 5.05 MiB

MiracleWhy, 2018-12-14:
i remove the build directory ,but when i run app,it show like this .
 
[image]

n1ru4l, 2018-12-14:
@MiracleWhy please check https://github.com/react-native-community/react-native-camera/issues/1844

MiracleWhy, 2018-12-16:
> @MiracleWhy please check #1844

tks,it might be my com.google.xxx is not the right version

PR #2001 (2018-12-17): fix: exclude android build artifacts
Closes #2000

TODO: Test if it breaks build as mentioned in https://github.com/react-native-community/react-native-camera/issues/2000#issuecomment-447328777
closed 2018-12-17 by n1ru4l