react-native-camera/react-native-camera #2008: net::ERR_FILE_NOT_FOUND when trying to fetch uri of TakePictureResponse

atrauzzi, 2018-12-17:
# Bug Report
**To Do First**
- [X] Did you try latest release?
- [X] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
Android

**Description/Current Behaviour**

When I run the following:

```typescript
        const picture = await this.camera.takePictureAsync({
            exif: true,
            doNotSave: false,
        });

        const fileRetrieve = await fetch(picture.uri);

        const data = await fileRetrieve.blob();
```

...the last line triggers the following exception (regardless of whether I have the debugger attached or not):

> Unhandled promise rejection TypeError: Network request failed

When I have the debugger attached and I'm inspecting network, I also see:

> GET file:///data/user/0/com.my.app/cache/Camera/6e499c20-6e3e-4233-becf-12a1d5e381aa.jpg net::ERR_FILE_NOT_FOUND

**Expected Behaviour**
I expect to have a successful fetch response where I can obtain a `Blob` reference to my file so that I can use binary apis rather than using memory intensive and less performant base64 strings.

**Steps to Reproduce**
The code above inside of even the most basic of camera setups is more than sufficient.

I'm not using Expo.

**Additionals**

I have the following permissions in my `AndroidManifest.xml`:

```xml
    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW" />
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
```

cc.
 - https://github.com/react-native-community/react-native-camera/issues/1361

n1ru4l, 2018-12-17:
What exactly do you want to do with the image?

atrauzzi, 2018-12-17:
I'd like to get a `Blob` object.  Initially it'll be for sending the file to a server, but we also have a few other operations in mind where it'll be nice to have the standard `Blob` representation (watermarking, encryption).

Biggest thing here is I want to avoid loading all the data into memory as a base64 encoded string.

atrauzzi, 2018-12-17:
Entering this in as a separate comment, but it appears as though I've been able to get access to the file thanks to [this comment](https://github.com/expo/expo/issues/2402#issuecomment-443726662).

Seems like there's a problem in react native with `fetch`...

closed 2018-12-17 by atrauzzi