react-native-camera/react-native-camera #2011: actually implement captureAudio

n1ru4l, 2018-12-18:
# Feature-Request

**Describe the Feature**
The `captureAudio` property on the `RNCamera` instance is currently only used for determining whether the `requestPermissions` code should ask for Android Audio Recording permissions or not.

I previously removed documentation for this property since I thought it is not implemented on `RNCamera`.

We should reimplement it with the following functionality:

- fallback value for recording if the `mute` option is not passed into `recordAsync`.

**Related Issues**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

arneson, 2019-01-02:
@n1ru4l 
In my opinion, after looking at it, I think it makes the most sense to auto-mute if you have not passed captureAudio: true to the component. Do you agree? If you do I'll put that together and create a PR. 

Secondly, are you saying that recording without audio is not currently supported and need to be implemented as well or does this issue only refer to the captureAudio/mute params thing?

n1ru4l, 2019-01-04:
@arneson What about not having set the `captureAudio` property and not using the `mute` option with `false`?

Recording without audio is currently supported. You will have to pass  the option `mute` with the value `false` to the `recordAsync` method.

arneson, 2019-01-07:
> What about not having set the captureAudio property and not using the mute option with false?

By not having set the captureAudio prop you mean changing the default of captureAudio to true? 

To summarize: 
1. Change captureAudio default to true
2. If captureAudio is set to false, disregard mute property and always pass mute:true to the camera?

n1ru4l, 2019-01-07:
@arneson Yeah that would be a solution without any breaking changes (since I consider recording audio by default but not asking for permissions as a bug). Are you interested in fixing this issue?

In the next breaking change release we could then deprecate and remove the `mute` property.

arneson, 2019-01-07:
@n1ru4l 
Yeah! Sure! I'll take a look. I have that basically done in my fork, just need to look through it once more.

n1ru4l, 2019-01-18:
Implemented in #2037 and #2048

closed 2019-01-18 by n1ru4l