react-native-camera/react-native-camera #2012: Make a Blob representation available (without having to go through URI)

atrauzzi, 2018-12-18:
# Feature-Request

**Describe the Feature**

When images are taken, the `TakePictureResponse` has options for `base64` and a `uri`.  Unfortunately, `Blob` support in react native is not in the best of shape currently (see related issue below).  This means that I have no way of obtaining a compatible `Blob` using the `fetch` API and `uri`.

I'd like to see another key added to `TakePictureResponse` called `blob`, `data` or `bytes` (anything along those lines, really).  This would be an object reference to a standard instance of `Blob` that could then be used by other JS libraries that leverage the standard.
A feature like this should also offer some performance improvements for any react native apps that use the camera:

- The hop to base64 will no longer be implied/required
- Bytes can be read in chunks for any JS code looking to do manipulations
- Bytes can be streamed directly from storage or between native APIs

**Possible Implementations**
I presume at the point where a native file reference is obtained, it would have to be mapped over to a standard `Blob` object and added to the `TakePictureResponse`.

**Related Issues**

https://github.com/facebook/react-native/issues/22681

n1ru4l, 2018-12-19:
Does this need to be solved in `react-native-camera`? 

Can't you create a `Blob` with the method described by you and the current workaround mentioned in https://github.com/expo/expo/issues/2402#issuecomment-443726662?

atrauzzi, 2018-12-19:
Nope, if you read my referenced issue, you'll see that fetch doesn't return a standard Blob.  It's broken and can't be used outside of their implementation. As in... If you pass that blob to anything that expects one, it just ends up being treated as an object.

I think there's merit either way for you to offer the blob, even if their blob implementation wasn't broken.  Blobs are native codes best shot at interoperability with other native libraries.  It's also going to be marginally faster than requiring people to resolve via filesystem.  You might even improve support for platforms that don't like to expose filesystems.

sibelius, 2018-12-19:
Blob is a great idea, feel free to send a pr to support them

n1ru4l, 2018-12-19:
I still not understand whats the issue with `react-native-camera` since it must be fixed in `react-native` core. And you could then create a Blob by using the uri property of the `recordAsync`/`takePictureAsync` result?

atrauzzi, 2018-12-19:
@n1ru4l - As I outline in the linked issue: The blob from `fetch` using `uri` is **broken**.  You can't even slice it.  It just returns an instance of itself of the same `size`.  You can't pass it to axios because it doesn't have `Blob` as its prototype. Simply put, it's not a Blob.  It's a react-native-not-blob.  So it can't be used with any library designed to use the standard ECMA blob APIs.

But a better reason for camera to offer a blob directly is that by doing so, you are removing the extra hop of having to go through the filesystem.  It's more performant and as an added bonus, opens up your API to support sources that don't necessarily expose a filesystem.

sibelius, 2018-12-19:
I think this should be solved on core first

let's reopen this when core is fixed

have you tested this example https://github.com/expo/react-native-blob-test?

closed 2018-12-19 by sibelius

atrauzzi, 2018-12-19:
Whether it's fixed in core or not probably won't affect your implementation of directly offering a standard blob.  This seems like something you could do right away and help people out greatly -- core is probably going to take a long time to fix this.

atrauzzi, 2018-12-19:
And like I said, I'm sorry, I literally don't have the background to implement it.

n1ru4l, 2018-12-19:
@atrauzzi I understand your frustration, but currently the best workaround is simply using fetch instead of axios as @sjchmiela mentioned in his comment (https://github.com/expo/expo/issues/2402#issuecomment-443726662). 

As @sibelius said we are happy to accept pull requests for this, but it does most likely make more sense to fix the in `react-native` core.

n1ru4l, 2018-12-19:
@atrauzzi If you are interested this is how I do fileuploads at the moment:

```js
function sendFile(presignedurl, file, onSuccess, onFail, onProgress) {
  // eslint-disable-next-line
  const xhr = new XMLHttpRequest();
  xhr.upload.addEventListener(
    `progress`,
    e => {
      // handle notifications about upload progress: e.loaded / e.total
      if (onProgress) {
        onProgress(e);
      }
    },
    false
  );
  xhr.onreadystatechange = function() {
    if (xhr.readyState === 4) {
      if (xhr.status === 200) {
        // Successfully uploaded the file.
        onSuccess(file.uri);
      } else {
        // The file could not be uploaded.
        const error = new Error(
          `Something went wrong... ${xhr.status} ${xhr.responseText}`
        );
        onFail(error);
      }
    }
  };
  xhr.open(`PUT`, presignedurl);
  xhr.send(file);
}

function upload(
  uri // the uri returned by recordAsync/takePictureAsync
) {
  return new Promise((resolve, reject) => {
    const file = {
      uri,
      name: "foobar.jpeg"
    };
    sendFile(data.requestFileUpload.uploadUrl, file, resolve, reject);
  });
}

```

atrauzzi, 2018-12-19:
@n1ru4l - If I want to do binary manipulation of the image, using fetch won't work.  Your solution of using `fetch` assumes all I want to do is send it over the network.  I can't chunk it, I can't do anything except feed it back to the same RN API that generated it.

Offering a binary representation without requiring people to find the file on the filesystem is absolutely a better option for you to offer.  It removes a layer of indirection and simplifies your API.  It allows this project to offer image data without having to assume a filesystem is there for the indirection to occur.

It's just good software design to remove a dependency in this case.  The whole reason why this project can't offer a standard blob is because you're using RN's `fetch` as a crutch.

atrauzzi, 2018-12-19:
This is unrelated.. But for what it's worth, I have a transfer working without using base64 data currently.  It's a complete workaround and some ugly one-off code, but as I mention in my last comment, I can't use chunking and I can't use axios.
(My organization has all of its network operations abstracted using an internal package built on axios.  We did this because axios builds on standard APIs -- when they aren't broken obviously.  Again, unrelated, but I just want you to know I've definitely gone as far as I can.)