react-native-camera/react-native-camera #2017: Github Bot that adds install command to pull requests

n1ru4l, 2018-12-19:
# Feature-Request

**Describe the Feature**
It would be helpful if we could have a bot that automatically posts the `yarn` or `npm` install command to a new pull request, so that bug reporters that get pointed to a pull request have no additional obstacle of testing the pull request (Many people do not know you can install a git npm package etc.).

@sibelius What do you think?

**Possible Implementations**
I have seen projects that do this. However at the moment I cannot remember the name of any 😅

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2018-12-19:
great, we can try to use github actions or hubot for this

n1ru4l, 2018-12-19:
@sibelius Do we have access to github actions?

sibelius, 2018-12-19:
not yet, https://twitter.com/sseraphini/status/1075362702233624577

closed 2019-06-08 by stale[bot]