react-native-camera/react-native-camera #2018: Migrate to ML Kit

sibelius, 2018-12-19:
# Feature-Request

**Describe the Feature**
Google Mobile Vision (GMV) is deprecated https://developers.google.com/vision/
We should migrate to ML Kit https://developers.google.com/ml-kit/

## How to implement this
- [ ] Add ML Kit for both android and ios
- [ ] Provide a migration path to use ML Kit instead of GMV
- [ ] Remove GMV from RNCamera

## Features that this would enable
- [ ] Image Labeling
- [ ] Barcode scanning
- [ ] Text Recognition
- [ ] Landmark detection
- [ ] Face detection

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

n1ru4l, 2018-12-19:
This would definetly be a big breaking change. I think we should setup a roadmap for version 2.0

sibelius, 2018-12-19:
sure, we could try to use github boards or milestone to keep track of v2.0

sibelius, 2019-04-18:
done by @dov11

fabriziobertoglio1987, 2020-04-02:
@cristianocca @sibelius @MateusAndrade would like to open a discussion about this and start working on MLKIT features soon.

as my understanding not all the MLKIT features are available in our library or they don't have complete documentation so I will assign myself this tasks and work on the following features as soon as possible:

1) [Image Labeling][1]
2) [landmark-detection][2]
3) [object detection and tracking][3]
4) [auto-ml-vision-edge][4]
5) [translation][5]
6) [custom ML][6]

I am also very interested in TensorFlow and will look at that api later. I am available Full-Time.
Thanks a lot.. 

[1]: https://developers.google.com/ml-kit/vision/image-labeling
[2]: https://developers.google.com/ml-kit/vision/landmark-detection
[3]: https://developers.google.com/ml-kit/vision/object-detection-and-tracking
[4]: https://developers.google.com/ml-kit/vision/auto-ml-vision-edge
[5]: https://developers.google.com/ml-kit/language/translation
[6]: https://developers.google.com/ml-kit/custom

sibelius, 2020-04-02:
I think each mlkit feature could be a different issue and pull request

ceefour, 2020-04-11:
Current docs: https://react-native-community.github.io/react-native-camera/docs/rncamera#ongooglevisionbarcodesdetected

closed 2020-06-17 by stale[bot]