react-native-camera/react-native-camera #2019: Version 2 Discussion

n1ru4l, 2018-12-19:
# Version 2 Discussion

Version 2 should be all about removing deprecated dependencies and APIs and fixing bad APIs.

Roadmap Overview: https://github.com/react-native-community/react-native-camera/projects/1

Please link all crucial issues that should be part of v2 in this Thread.

n1ru4l, 2018-12-19:
@sibelius we can develop v2.0 in a new branch and use semantic-release to create v2.0 pre-releases (https://github.com/semantic-release/semantic-release/issues/563#issuecomment-448538232).

n1ru4l, 2019-01-09:
# Idea: Do not record audio by default

Recording audio requires additional permissions. A developer that only wants to scan a qr code does not require recording audio.

Therefore it would be best if by default audio recording is disabled.

Only if the developer added the `captureAudio` property to the component (and permissions are granted). Also I think we should differentiate between the both cases:

1. `Camera allowed` and `Audio not allowed` 
2. `Camera allowed` and `Audio allowed`

In the first case the camera view is still  functional and the developer should decide whether the app user is forced to also allow audio permissions or is allowed to record without video.

Related: #2041

sfgghh, 2019-01-21:
Leave

n1ru4l, 2019-01-23:
## Idea: Migrate to TypeScript and provide flow type definitions

TypeScript is generally more widespread than flow. There is currently no check that ensures the validity of flow type annotations of `react-native-camera`.

n1ru4l, 2019-01-23:
## Idea: Use depfu for handling dependencies

https://github.com/marketplace/depfu

n1ru4l, 2019-02-18:
## Idea: Remove prop-types
Having TypeScript and flow type definitions should me more than enough.

sraka1, 2019-03-05:
@n1ru4l https://github.com/react-native-community/react-native-camera/projects/1#card-15903168 Regarding "[Android] remove Camera1 API and replace it completely with Camera2 API" doesn't that bump the minSdkVersion to 21 (meaning Android 5.0 or higher)? More than 10% of global Android users are still on a version below 5.0.

Bardiamist, 2019-03-21:
### Feedback
My project required to scan qr-code. It was implemented by another developer with `react-native-camera@1.13.1`.

Today I tried to migrate to V2 but faced the following problems:
1. In README is not clear what is `Barcode Detection`. It `Face Detection` or `Text Recognition` or should works without additional steps on iOS. [I found what barcode recognition for iOS not impplemented not migrated to MLKit](https://github.com/react-native-community/react-native-camera/pull/2075#issuecomment-462522320) but `onBarCodeRead` not works after update without additional steps and I not sure what will work because [`googleVisionBarcodeMode` marked as android only](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md#android-ongooglevisionbarcodesdetected).
2. `Text recognition is currently available only via CocoaPods Path.` It bad. I do not want to use Pods [while CocoaPods is not in the project template](https://github.com/facebook/react-native/pull/23563).
3. It required firebase. Before `Barcode Detection` worked without firebase. It is not clear how friendly `react-native-camera` with `react-native-firebase`. **Nowhere is there an explanation why it is required MLKit now.**

**I spent time searching for these answers and I had to roll back to `react-native-camera@1.13.1`.**

Very important to update README and docs.
1. Add desription about `Barcode Detection` in [README](https://github.com/react-native-community/react-native-camera#face-detection-or-text-recognition-steps).
2. Depricate [`onBarCodeRead`](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md#onbarcoderead) or use one property for both platforms.
3. Why [Permissions section not in `Geting started`](https://github.com/react-native-community/react-native-camera#permissions).

ThaJay, 2019-05-15:
I just hope react-native-camera v1 will keep receiving support for the foreseeable future as we just need a viewfinder and barcode / qr code scanner.

n1ru4l, 2019-06-06:
Version 2 was released without addressing many of these issues... Going to close this. If anyone wants to address any of the mentioned issues please send a pull request.

closed 2019-06-06 by n1ru4l