react-native-camera/react-native-camera #2022: mute option is not properly documented or badly implemented.

ricbermo, 2018-12-20:
# Bug Report
**To Do First**
- [ ] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

RNCamera

**Platforms**

Android
iOS

**Versions**

- Android:
- iOS:
- react-native-camera: 1.6.3
- react-native: 0.57.7
- react: 16.6.1

**Description/Current Behaviour**
It looks like the the description of `mute` is not acurate.
mute (any value). If this flag is given in the option with any value, the video to be recorded will be mute. If nothing is specified, video will NOT be muted.

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

n1ru4l, 2018-12-21:
We have currently definitely some issues regarding audio and video.

Are you sure not passing  `mute: false` to `recordAsync` will result in no audio?

On iOS I can reproduce the issue for sure. 

Can you please test https://github.com/react-native-community/react-native-camera/pull/2024 and report there if it resolves this issue that you are experiencing.

ricbermo, 2018-12-21:
hey, @n1ru4l for sure. I'll do that once I have some time.
This is what I have as `options`.
```javascript
const options = {
  mirrorVideo: false,
  maxDuration: 30,
  mute: false,
};
```

closed 2019-06-08 by stale[bot]