react-native-camera/react-native-camera #2023: Barcode scanner bounds units

KlemenS189, 2018-12-21:
## Question
I have a question about units that are used in the bounds.
Since the plugin doesn't offer the possibility to customize the area in which to scan, I want to restrict the area with bounds. The problem is that I can't correlate the units of the returned bounds with any other units used in react-native.

## Example
I am using LG G6.
To get the dpi units of the screen I used Dimension.get('window').height & Dimension.get('window').width. I get the following results:

1. Dimension.get('window').width = 360
2. Dimension.get('window').height = 720

When transforming them to px units, I get:

1. Width = 1440px
2. Height = 2880px

I tried to place the barcode to the left of the screen I get the following bounds:
{y: "640.0", x: "98.5"} {y: "640.0", x: "419.5"} 
First X coordinate should be around 0?

Then I tried to place the barcode to the right of the screen I get the following bounds:
{y: "640.0", x: "537.0"} {y: "640.0", x: "844.5"}

These units are behaving quite weird.
Does anyone know, how I can convert them to dpi or px?

crobinson42, 2019-07-18:
@KlemenS189 I'm facing this issue right now - any advice on how to capture barcode reading within certain area/bounds of the camera image?

MOST of the time the barcode is read in the boxed area but there are times when the barcode below it is read instead and this is frustrating :-|

[image]

closed 2019-09-23 by stale[bot]

helciofranco, 2020-03-23:
I'm facing the same issue. The barcode is on the edge-left but Android is recognizing as something like x > 100.