react-native-camera/react-native-camera #2027: A problem occurred configuring project ':react-native-camera'.

mopilo, 2018-12-25:
# Bug Report
**To Do First**
- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera: 1.2.0
- react-native: 0.56.0
- react: 16.4.1

**Description/Current Behaviour**

**Expected Behaviour**
Need the app to build

**Steps to Reproduce**
Added google() to my build.gradle file

**Additionals**
> Could not resolve all files for configuration ':react-native-camera:classpath'.
   > Could not find manifest-merger.jar (com.android.tools.build:manifest-merger:26.0.0).
     Searched in the following locations:
         https://jcenter.bintray.com/com/android/tools/build/manifest-merger/26.0.0/manifest-merger-26.0.0.jar
   > Could not find ddmlib.jar (com.android.tools.ddms:ddmlib:26.0.0).
     Searched in the following locations:
         https://jcenter.bintray.com/com/android/tools/ddms/ddmlib/26.0.0/ddmlib-26.0.0.jar
   > Could not find dvlib.jar (com.android.tools:dvlib:26.0.0).
     Searched in the following locations:
         https://jcenter.bintray.com/com/android/tools/dvlib/26.0.0/dvlib-26.0.0.jar
   > Could not find common.jar (com.android.tools:common:26.0.0).
     Searched in the following locations:
         https://jcenter.bintray.com/com/android/tools/common/26.0.0/common-26.0.0.jar

dmitryn, 2018-12-25:
I also have this issue, here is my build.gradle:
```
buildscript {
    repositories {
        flatDir { dirs "libs", "${rootDir}/app/libs" }
        google()
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.0.1'
        classpath 'com.google.gms:google-services:4.1.0'
        classpath 'de.undercouch:gradle-download-task:3.1.2'
    }
}

allprojects {
    repositories {
        mavenLocal()
        google()
        jcenter()
        maven { url "$rootDir/../node_modules/react-native/android" }
        flatDir { dirs "libs", "${rootDir}/app/libs" }
        maven { url "https://jitpack.io" }
    }
}
```

* react-native-camera 1.1.5
* react-native 0.56

mopilo, 2018-12-26:
were you able to solve it??

mopilo, 2018-12-26:
This what fixed it for me.

https://github.com/react-native-community/react-native-image-picker/issues/999#issuecomment-445465847

dmitryn, 2018-12-26:
>were you able to solve it??

@mopilo upgrade to `1.6.4` solved this issue for me

dmitryn, 2018-12-26:
Here are  different `build.gradle` for each version:

https://github.com/react-native-community/react-native-camera/blob/v1.1.5/android/build.gradle
https://github.com/react-native-community/react-native-camera/blob/v1.6.4/android/build.gradle

dmitryn, 2018-12-26:
Solution is to move `jcenter()` down the list in `build.gradle`:

```
  repositories {
    google()
    maven {
      url 'https://maven.google.com'
    }
    jcenter()
  }
```

n1ru4l, 2018-12-26:
As pointed out by @dmitryn this issue is solved when using the newest version of `react-native-camera` (`1..6.4`).

closed 2018-12-26 by n1ru4l