react-native-camera/react-native-camera #2029: no audio

yernandus, 2018-12-27:
# Bug Report
**To Do First**
- [x] Did you try latest release? 
- [ ] Did you try master?
- [x ] Did you look for existing matching issues?

**Related Modules**

RNCamera

**Platforms**

iOS

**Versions**

- iOS:latest
- react-native-camera:1.6.4
- react-native:0.57.5
- react:16.6.1

**Description/Current Behaviour**

video is recorded without audio

**Expected Behaviour**

recorded videos should have audio
**Steps to Reproduce**

record an video on ios. 

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

yernandus, 2018-12-27:
solved by passing { muted: false } to recordAsync. The documentation is misleading

n1ru4l, 2018-12-29:
@yernandus Can you please test if https://github.com/react-native-community/react-native-camera/pull/2024 fixes your issue?

n1ru4l, 2019-01-08:
:tada: This issue has been resolved in version 1.6.5 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.6.5)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

closed 2019-01-08 by n1ru4l

n1ru4l, 2019-01-08:
:tada: This issue has been resolved in version 1.6.6 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.6.6)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #2024 (2019-01-08): fix(ios): workaround accidental breaking change
Introduced here: https://github.com/react-native-community/react-native-camera/commit/401c485#diff-deff8cbb3789a329a4a17e2e3bd13bceR534

However when you set `mute: true` it will cause flickering. This is only to fix the breaking change. The "final" solution will be the `captureAudio` property (#2011).

Fixes #2029
commit 4bdcbd37 (2019-01-08): fix(ios): workaround accidental breaking change (#2024)

v1.6.0 introduced an accidental breaking change which resulted in videos being recorded without audio.

Closes #2029