react-native-camera/react-native-camera #2031: Unable to use play-services-vision:15.0.2

laukaichung, 2018-12-28:
I've installed both react-native-firebase and react-native-camera. The camera was fine when play-services -vision was stuck at `12.0.1`, but I just ran into this error (**Error updating property googleVisionBarcodeDetectorEnable**)  https://github.com/react-native-community/react-native-camera/issues/1844 that requires an upgrade to `15.0.2`.

  It looks like there is a Google Play Services and Firebase conflicts when `play-services-vision` is bumped up to `15.0.2` from `12.0.1`:

````
Dependency failing: com.google.android.gms:play-services-flags:15.0.1 -> com.google.android.gms:play-services-basement@[
  15.0.1], but play-services-basement version was 16.0.1.
  
  The following dependencies are project dependencies that are direct or have transitive dependencies that lead to the art
  ifact with the issue.
  -- Project 'app' depends onto com.google.firebase:firebase-messaging@17.3.4
  -- Project 'app' depends onto com.google.android.gms:play-services-base@16.0.1
  -- Project 'app' depends onto com.google.firebase:firebase-core@16.0.6
  -- Project 'app' depends onto com.google.android.gms:play-services-vision@15.0.2
````

I've tried `com.google.android.gms:play-services-vision@16.2.0` but it gave me `exceed 64k methods` error. 
Bumping up to 17.0.2 would cause a version conflict from `com.google.android.gms:play-services-basement`. 

Is there anyone who use both react-native-firebase and react-native camera? Can you tell me how to solve this version conflict problem?

Here is the dependencies in android/app/build.gradle
````
dependencies {

    implementation (project(':react-native-camera')) {
      exclude group: "com.google.android.gms"
      implementation "com.android.support:exifinterface:${rootProject.ext.supportLibVersion}"
      implementation ('com.google.android.gms:play-services-vision:12.0.1') {
        force = true
      }
    }
    implementation project(':react-native-gesture-handler')
    implementation project(':react-native-webview')
    implementation project(':react-native-fast-image')
    implementation project(':react-native-google-signin')
    implementation project(':react-native-firebase')
    implementation 'com.google.firebase:firebase-core:16.0.6'
    implementation ('com.google.android.gms:play-services-base:16.0.1')
    implementation 'com.google.firebase:firebase-messaging:17.3.4'
    implementation('com.crashlytics.sdk.android:crashlytics:2.9.5@aar') {
        transitive = true
    }
    implementation(project(':react-native-google-signin')) {
        exclude group: "com.google.android.gms" // very important
    }
    implementation fileTree(include: ['*.jar'], dir: 'libs')
    implementation "com.android.support:appcompat-v7:${rootProject.ext.supportLibVersion}"
    implementation 'com.facebook.react:react-native:+'
    implementation project(':react-native-sqlite-storage')
    implementation 'me.leolin:ShortcutBadger:1.1.21@aar'
    implementation 'com.facebook.fresco:animated-gif:1.10.0'
}
````

Ext in android/build.gradle
````
ext {
    buildToolsVersion = "28.0.3"
    minSdkVersion = 19
    compileSdkVersion = 28
    targetSdkVersion = 28
    supportLibVersion = "27.1.1"
}

````

n1ru4l, 2018-12-29:
Please read https://developer.android.com/studio/build/multidex

n1ru4l, 2019-01-30:
@laukaichung It seems like you did not read the link I shared... https://developer.android.com/studio/build/multidex

closed 2019-06-08 by stale[bot]