react-native-camera/react-native-camera #2034: Is there any way to capture image of scanned bar code?

ananth10, 2019-01-04:
Hello Developers!!

I am using this library  to scan barcode . it is working perfectly. but I need to capture image of barcode.

I have came to know about this scanner app [](https://play.google.com/store/apps/details?id=com.gamma.scan&hl=en). it would scan barcode and display it in next screen. I need exact like this.
please provide me idea . I will really appreciate it.

Thanks.

n1ru4l, 2019-01-08:
This is currently not possible. However we are always open to contributions. So if you want to implement this feature feel free to send a pull request

msageryd, 2019-01-17:
@ananth10 Do you need to display an exact copy of the captured image? Or do you simply need to display a barcode that matches the value of the captured barcode?

I had the same problem wit QR codes. I didn't want to store jpg-files of captures qr codes. Instead I store the value of the qr code and when I need to display the qr code I render a new one with react-native-qr-code-svg.

I suspect there exists libraries for generating bar codes as well.

closed 2019-06-08 by stale[bot]

anupbangale, 2023-03-10:
> Hello Developers!!
>
> I am using this library to scan barcode . it is working perfectly. but I need to capture image of barcode.
>
> I have came to know about this scanner app [...]

Did you get any working solution on this?