react-native-camera/react-native-camera #2035: overlay to image before saveToCameraRoll

jasan-s, 2019-01-06:
# Question
Given a CGRect(x,y, width, height) I wanted to selectively only add pixelate effect or black color fill to only that part of an image. i.e pixelate rectangle over users face. I want to be able to do this on the device instead of on a server(most image manipulation tools like [Jimp](https://github.com/oliver-moran/jimp) don't work in react-native ). Is it possible to do this purely in react-native ? I have searched and can't seem to find a solution. 

**Tags**
image-manipulation,  overlay, composition, image-editing

nubwang, 2019-01-07:
为啥不能用哇，刚init一个项目，加了这个插件，安装了link了，Could not resolve all files for configuration ':react-native-camera:releaseCompileClasspath'.
> Could not resolve com.facebook.react:react-native:+.
  Required by:
      project :react-native-camera
   > Could not resolve com.facebook.react:react-native:+.
      > Failed to list versions for com.facebook.react:react-native.
         > Unable to load Maven meta-data from https://jitpack.io/com/facebook/react/react-native/maven-metadata.xml.
            > Could not get resource 'https://jitpack.io/com/facebook/react/react-native/maven-metadata.xml'.
               > Could not GET 'https://jitpack.io/com/facebook/react/react-native/maven-metadata.xml'.
                  > Read timed out

n1ru4l, 2019-01-08:
Duplicate of https://github.com/gre/react-native-view-shot/issues/85

I do not know how `react-native-view-shot` works, so I cannot help you with that. If you have time to investigate further (and this issue can be resolved by adding changes to `react-native-camera`) I encourage you to provide a pull request.

As a **workaround** you could show the taken picture with an `ImageBackground` component and use `react-native-view-shot` on the `ImageBackground` component.

Angelk90, 2019-02-04:
@n1ru4l : In reference to the issues #2092, that have been closed without being able to reply to the message.

I do not understand the answer that has been given.
Which means: This is out of scope of this library ...

I posted the code with an example, and I asked how I could do something, saving by applying a filter for example, I did not ask things that have nothing to do with the module.

n1ru4l, 2019-02-04:
@Angelk90 It simply means this is not an issue related to the functionality of react-native-camera. Issues are intended for requesting new features or reporting bugs.

Questions should be asked on StackOverflow.

I do not have time to provide examples for things not covered by the core features. Therefore I closed your issue.

Also the example you provided is for the expo-sdk which is not react-native-camera.

Angelk90, 2019-02-04:
@n1ru4l : Are you telling me that the request to add a `SurfaceView `function on the camera, for example, in order to do what I asked, for example to apply a filter through a kernel, is beyond the requests that can be made here?

[image]

n1ru4l, 2019-02-04:
@Angelk90 Nope, I never said that. Feel free. It would even be better if you could implement it.

closed 2019-06-08 by stale[bot]

fera765, 2020-10-14:
has anyone managed to solve this problem?

emrahc, 2020-12-16:
@Angelk90 were you able to do it?Saving picture with facial square?