react-native-camera/react-native-camera #2040: recordAsync() does not fire a second time

keithj0nes, 2019-01-08:
Bug Report

**Related Modules**

RNCamera

**Platforms**

iOS

**Versions**

- Android:
- iOS:  11.2
- react-native-camera: 1.2.0
- react-native: 0.56.0
- react: 16.4.1
- react-navigation: 2.11.2

**Description/Current Behaviour**

home screen -> camera screen -> details screen

My UI is set up to where when the video is done recording, it navigates to the details screen where I can preview the video. If I go back to the camera and try to record a video again, the `this.camera.recordAsync()` function doesn't get called that second time (it only gets called the first time).

If I go back to the home screen before the camera screen, and navigate to the camera screen again, the video resets and I can take a second video - ideally the camera unmounting causes the `this.camera.recordAsync()` to be able to be fired again.

**Expected Behaviour**

I should be able to record a second video after navigating back from the details screen without having to have my users go to navigate back twice to the home screen in order to take a new video after the first video has been recorded.

**Steps to Reproduce**

```javascript
takeVideo = async () => {
  if(this.camera){
    this.setState({isRecording: true})

    const options = {
      orientation: 'auto', 
      quality: RNCamera.Constants.VideoQuality["480p"], 
      maxDuration: 5,
    }

    this.camera.recordAsync(options).then((data) => {  // fires first time, does not fire second time
      
      this.setState({videoPath: data.uri, isRecording: false})
      this.props.navigation.navigate('captureDetails', {filePath: data.uri, type: 'video'})  // react navigation

    })
    .catch(err => console.log(err, 'errorrrrrr'));
  }
}

stopVideo = () => {
  this.setState({isRecording: false}, () => {
    this.camera.stopRecording();
  });
}
```

Any ideas or suggestions?

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

yogisalomo, 2019-02-13:
I'm having a similar problem of executing _recordyAsync_ multiple times like this issue. The original issue solved it with some navigation event, however in my case, I simply just want to call the _recordyAsync_ function again in the same component.

Every time I call _recordyAsync_ for the second time, I faced this `[Error: Starting video recording failed. Another recording might be in progress.]` error/rejection. And I'm pretty sure the previous _recordAsync_ promise is already resolved because I've obtained the URI of the result.

Briefly, my steps were:
- Call _recordyAsync_
- Call _camera.stopRecording()_
- Call _recordyAsync_ again

Here are the output when I tested (I console log every time I call a RNCamera function):

> 02-13 11:58:19.981 19873 20012 I ReactNativeJS: **Record Async Called for the First Time**
> 02-13 11:58:23.578 19873 20012 I ReactNativeJS: Current status recording
> 02-13 11:58:23.800 19873 20012 I ReactNativeJS: **Stop Recording is Called**
> 02-13 11:58:24.321 19873 20012 I ReactNativeJS: 'RecordAsync **Promise is resolved with result data**:', { uri: 'file:///data/user/0/com.superd/cache/Camera/dbb2083d-bf12-4002-9ecf-fc7cb3d271c2.mp4',
> 02-13 11:58:24.321 19873 20012 I ReactNativeJS:   deviceOrientation: 1,
> 02-13 11:58:24.321 19873 20012 I ReactNativeJS:   videoOrientation: 1 }
> 02-13 11:58:29.549 19873 20012 I ReactNativeJS: Record button is pressed during PAUSED state
> 02-13 11:58:29.801 19873 20012 I ReactNativeJS: **Record Async Called for the Second Time**
> 02-13 11:58:29.862 19873 20012 I ReactNativeJS: **'resume recording error:', { [Error: Starting video recording failed. Another recording might be in progress.]**
> 02-13 11:58:29.862 19873 20012 I ReactNativeJS:   framesToPop: 1,
> 02-13 11:58:29.862 19873 20012 I ReactNativeJS:   code: 'E_RECORDING_FAILED',
> 02-13 11:58:29.862 19873 20012 I ReactNativeJS:   line: 25424,
> 02-13 11:58:29.862 19873 20012 I ReactNativeJS:   column: 31,
> 02-13 11:58:29.862 19873 20012 I ReactNativeJS:   sourceURL: 'http://localhost:8081/index.delta?platform=android&dev=true&minify=false' }

I believe this should not happen as it is quite a normal case for multiple times execution of a camera recording. Here are several methods that I have tried to solve this:
- Not rendering the RNCamera component for a moment and rendering again, same error still appears
- Try to force rejecting the promise from previous recordAsync, still not working

I didn't put my code snippet as it was still very basic. however if it's needed to analyze more of the problem, I will gladly provide. I have tried to delve in to the RNCamera code for a bit but I couldn't find anything.

If this is already solved/discussed, or if there is some workaround, or maybe there are mistakes on my part that caused this error, please give me some pointer.  Any help is greatly appreciated. And if this is an actual problem in RNCamera, I would love to discuss more and try to help with my limited ability.

Thank you.

furkancelik, 2019-03-25:
Is there a solution for v2.*?

does not record video when I change the camera type..

closed 2019-06-07 by stale[bot]