react-native-camera/react-native-camera #2042: Does RCTCamera support pinch to zoom on iOS?

mtr1012, 2019-01-09:
I'm using RCTCamera as camera component on my project.
But I can't pinch to zoom on iOS but on Android, everything works fine.

My question is: How can I support pinch to zoom on iOS? 

Many thanks

n1ru4l, 2019-01-09:
`RCTCamera` is deprecated.

Pinch zoom is supported in user land. If you come up with a recipe we might add it .

Checkout the documentation for https://facebook.github.io/react-native/docs/gesture-responder-system.html

mtr1012, 2019-01-09:
> `RCTCamera` is deprecated.
> 
> Pinch zoom is supported in userland. If you come up with a recipe we might add it.
> 
> Check out the documentation for https://facebook.github.io/react-native/docs/gesture-responder-system.html

I want to support pinch to zoom by myself but RCTCamera doesn't have "zoom" prop so I can't set it by handling pinch gesture. So the only way I can have the pinch to zoom feature is migrate to RNCamera ?

Thanks for reply me @n1ru4l

n1ru4l, 2019-01-09:
@mtr1012 Yes, `RCTCamera` will be removed in the next major release.

closed 2019-01-09 by n1ru4l