react-native-camera/react-native-camera #2043: What version of play services should I use? (with react-native-firebase)

arapocket, 2019-01-10:
# Question

**Ask your Question**

What is the latest version of play services that includes play-services-vision?

I am trying to build my project with play-services 16.0.1 for react-native-firebase but I can see that there is no play-services-vision:16.0.1

lFaustus, 2019-01-14:
i used the play-services-vision:16.2.0
```
implementation 'com.google.android.gms:play-services-vision:16.2.0'
implementation (project(':react-native-camera')){
        exclude group: 'com.google.android.gms', module: 'play-services-vision'
}
```

MarkRunWu, 2019-01-14:
I found there're a property `googlePlayServicesVersion` to specify which version it would load.
But since the version would be missing in other 'com.google.android.gms:' module. 
May renaming the property name into `playServicesVersion` ?

arapocket, 2019-01-14:
@lFaustus Thanks for the reply. Unfortunately I am also using location and now that's complaining about a non-existent version.

@MarkRunWu I didn't know googlePlayServicesVersion starts from 16.x.x. Good to know. Unfortunately I was already using playServicesVersion.

closed 2019-06-08 by stale[bot]