react-native-camera/react-native-camera #2046: [Android] `useCamera2API` and barcode scanning isn't working

vinayan3, 2019-01-10:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

**Platforms**
Android

**Versions**
- Android:
  * Nexus 5X @ Android 8.1.0
  * Samsung S6 Edge(SMG-925A)  @ Android 7.0
- react-native-camera: 1.8.0
- react-native: 16.6.3
- react:  0.57.8

**Description/Current Behavior**
Camera is setup as
```
      <RNCamera
        ref={ref => {
          this.camera = ref;
        }}
        style={{
          flex: 1,

        }}
        useCamera2Api={true}
        permissionDialogTitle={'Permission to use camera'}
        permissionDialogMessage={'We need your permission to use your camera phone'}
        onBarCodeRead={this.onBarcode}
     / >
```

**
When pointing the camera at the barcode with the `Camera2API`  enabled there are no detected barcodes.

**Expected Behaviour**
The barcodes should be detected.

**Steps to Reproduce**
1. Set a `onBarCodeRead` to a function
2. Set `useCamera2Api={true}

**Additionals**
I have spun out  the example camera app  into a new example repo with a project created by  `react-native init`. 

For ease of reproducing this bug I've added in buttons which let you toggle which `CameraApi` is used. When you toggle the API or the barcode detection the camera will be unmounted, 5 second wait, and remounted. 
[image]

matt-oakes, 2019-01-11:
This is likely caused by the same issue as #1714

AgingChan, 2019-01-17:
I have encountered a similar issue. The only difference is that the set `useCamera2Api` to `true` will result in the `Face Detection` not working.

n1ru4l, 2019-01-18:
:tada: This issue has been resolved in version 1.8.1 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.8.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

sraka1, 2019-03-07:
@n1ru4l Barcode scanning still does not work with useCamera2Api. Please re-open this issue.

agrass, 2019-05-28:
Yes, I tested in 2.3.0 and useCamera2Api, and still does not working the barcode scanning.

MateusAndrade, 2019-05-29:
> Yes, I tested in 2.3.0 and useCamera2Api, and still does not working the barcode scanning.

I Will give a look at this as soon as possible, but you can try the latest version of rn-camera?

agrass, 2019-06-10:
It worked with last version of rn-camera. Not sure, but it seems that it take more time in read the code compared with Camera1.

sibelius, 2019-06-10:
can you measure time of the operation?

where is the most time going?

JessePreiner, 2019-08-09:
Hi, Mr. Bot. Definitely don't close, as the only reason it is stale is because it has not been addressed. The problem didn't fix itself.

zelardiel, 2019-10-10:
Still a problem mr bot. See referenced issue.

PR #2050 (2019-01-18): fix(android): use the correct camera orientation to fix image detection
### The issue
[This code](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java#L278-L284) in the library is taking in the `rotation` from the `CameraView` and converting it to the image rotation. It is, however, making some assumptions about the camera orientation which is why it is failing on some devices.

The `rotation` from the `CameraView` callback corresponds to the display orientation. Portrait is *usually* `0`, however, that's [... cut]
closed 2019-12-16 by stale[bot]