react-native-camera/react-native-camera #2047: Black screen at first render and null object reference  on android

shaktalles, 2019-01-10:
# Bug Report
**To Do First**
- [ x ] Did you try latest release?
- [ x ] Did you try master?
- [ x ] Did you look for existing matching issues?

**Related Modules**

RNCamera

**Platforms**

Android

**Versions**

- Android: 6.0.1
- iOS:
- react-native-camera: 1.8.0
- react-native: 0.57.8
- react: 16.6.3

**Description/Current Behaviour**

After creating a new app: react-native init appexample,
In the app folder, install the package: npm install react-native-camera --save,
And do automatic link: react-native link.
However, when running the app on device, I get a black screen instead of the camera preview.
Upon clicking on SNAP, I get:

"Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference"
getDouble
ReadableNativeArray.java:114

**Expected Behaviour**

Be able to see camera preview and take picture.

**Steps to Reproduce**

As above.

**Additionals**

Got same issue on other apps, I have been reading and trying every possible solution and iteration for the last two days.

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

GaneshSaiKumar, 2019-01-11:
@n1ru4l  same issue here

syntony, 2019-01-11:
same issue, thanks  @shaktalles for advice, downgrading solved this problem

n1ru4l, 2019-01-11:
Sorry for the inconveniences. Could you please try if https://github.com/react-native-community/react-native-camera/pull/2048 resolves your issue?

shaktalles, 2019-01-11:
hi @n1ru4l, I tried #2048 PR, now it gives a black screen, but with the loading spinner spinning forever. 
I will make more tests next week.

n1ru4l, 2019-01-14:
@shaktalles Do you only grant `CAMERA` permissions or also `RECORD_AUDIO` permissions?

n1ru4l, 2019-01-14:
@shaktalles Could you please provide a reproduction repository?

luissmg, 2019-01-15:
@n1ru4l I think this is because `react-native-camera` is always showing up the `notAuthorizedView` and not checking properly for the permission?
In my case this is happening, even after giving permission. I am only using the camera to take photos, no video.

luissmg, 2019-01-15:
Soooooooooo I was looking into the lib's code... This is what I found.
In [this](https://github.com/react-native-community/react-native-camera/pull/2037) PR the `captureAudio` is set to `true` by default. Yes, this fixes a lot of stuff for some people, but also ruins for others like me that are not using audio.

Why did this break?
In [this](https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L418) line, we take the props value. Before it was `false`, now is `true`. Then, in [here](https://github.com/react-native-community/react-native-camera/blob/master/src/handlePermissions.js#L20), before, when the value was `false`, it entered in the condition and returned whatever. Now, it does not enter in that condition.

Therefore, we, people used to have `captureAudio` has `false` by default, must set it to `false` by hand.

This fixed it for me using the most recent version (`1.8.0`). Please try this @shaktalles @GaneshSaiKumar @syntony 
@n1ru4l Please correct me if I am wrong!

asanz8, 2019-01-18:
@luissmg Setting ```captureAudio={false}``` did the job, using version 1.8.0

n1ru4l, 2019-01-18:
@asanz8 You could help by checking if #2048 solves this issue without having to set the `captureAudio` property to `false`.

luissmg, 2019-01-18:
@n1ru4l I will install your fork from `feature-audio-permissions` branch today and will let you know how it ended up

luissmg, 2019-01-18:
Already tested by @asanz8!

n1ru4l, 2019-01-18:
:tada: This issue has been resolved in version 1.9.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.9.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

closed 2019-01-18 by n1ru4l

kent2508, 2019-08-20:
> Soooooooooo I was looking into the lib's code... This is what I found.
> In [this](https://github.com/react-native-community/react-native-camera/pull/2037) PR the `captureAudio` is set to `true` by [...]

you saved my day bro :D

PR #2048 (2019-01-18): feat: allow camera scene when audio permissions are denied
[@n1ru4l wrote](https://github.com/react-native-community/react-native-camera/issues/2019#issuecomment-452688195):
> Furthermore I think we should differentiate between the both cases:
>
> - `Camera allowed` and `Audio not allowed`
> - `Camera allowed` and `Audio allowed`
>
> In the first case the camera view is still functional and the developer should decide whether the app user is forced to also allow audio permissions or is allowed to record without video.

**What this PR does:**

- Prevent the camera view from **not being rendered [... cut]
commit 22533ed8 (2019-01-18): feat: allow camera scene when audio permissions are denied (#2048), Fixes #2047, Fixes #2051