react-native-camera/react-native-camera #2051: RNCamera **will** crash on iOS if all three Privacy keys are not included in Info.plist

dubert, 2019-01-14:
# Bug Report

Upon reading the README I came across:

```
<key>NSCameraUsageDescription</key>
<string>Your message to user when the camera is accessed for the first time</string>

<!-- Include this only if you are planning to use the camera roll -->
<key>NSPhotoLibraryUsageDescription</key>
<string>Your message to user when the photo library is accessed for the first time</string>

<!-- Include this only if you are planning to use the microphone for video recording -->
<key>NSMicrophoneUsageDescription</key>
<string>Your message to user when the microphone is accessed for the first time</string>
```

Naive me, I wasn't planning on doing any recording, so I didn't include `NSMicrophoneUsageDescription`. My app kept crashing on render and I went digging down a number of rabbit holes. Frustrated and I went back to square one and added all three, and boom, it works. 

I would **highly** recommend removing the comments that suggest the last two keys are optional. In my experience they most certainly were not optional.

dubert, 2019-01-14:
Update: I am somewhat to blame. This only happens if `captureAudio` is not set to `false` which I did not set. Part of what stumped me was that `Camera` was working, but `RNCamera` was not. But this is because `Camera` defaults `captureAudio` to `false`, but `RNCamera` defaults it to `true`.

micchyboy237, 2019-01-15:
But how do we set it to false? I use this library because of react-native-qr-scanner, and it crashes when there is no microphone permission

n1ru4l, 2019-01-15:
@dubert I am working on making microphone permissions optional. Currently my time is limited so if someone could test #2048 and review it would help me massively.

Fsarmento, 2019-01-16:
I update yesterday react-native-camera for the latest version and my app started crashing. [This](https://github.com/react-native-community/react-native-camera/commit/2940faf8710d321f790419c3458a7eaf9127b93e#diff-6ee136227b7a6fdd96b7a213d5648f4eR261) commit from 8 days ago changed the default `captureAudio` from false to true.

n1ru4l, 2019-01-16:
@Fsarmento Could you please check if #2048 resolves that crash issue when `captureAudio` is not set?

n1ru4l, 2019-01-18:
:tada: This issue has been resolved in version 1.9.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.9.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

closed 2019-01-18 by n1ru4l

chj-damon, 2020-05-27:
> I update yesterday react-native-camera for the latest version and my app started crashing. [This](https://github.com/react-native-community/react-native-camera/commit/2940faf8710d321f790419c3458a7eaf9127b93e#diff-6ee136227b7a6fdd96b7a213d5648f4eR261) commit from 8 days ago changed the default `captureAudio` from false to true.
> To avoid the error (if you only use the camera w/o audio), set `captureAudio={false}`
> 
> Error on Xcode:
> `019-01-16 09:59:13.613561+0000 peeqDaycare[1965:797218] [access] This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist must contain an NSMicrophoneUsageDescription key with a string value explaining to the user how the app uses this data.`

Thanks a lot!

PR #2048 (2019-01-18): feat: allow camera scene when audio permissions are denied
[@n1ru4l wrote](https://github.com/react-native-community/react-native-camera/issues/2019#issuecomment-452688195):
> Furthermore I think we should differentiate between the both cases:
>
> - `Camera allowed` and `Audio not allowed`
> - `Camera allowed` and `Audio allowed`
>
> In the first case the camera view is still functional and the developer should decide whether the app user is forced to also allow audio permissions or is allowed to record without video.

**What this PR does:**

- Prevent the camera view from **not being rendered [... cut]
commit 22533ed8 (2019-01-18): feat: allow camera scene when audio permissions are denied (#2048), Fixes #2047, Fixes #2051