react-native-camera/react-native-camera #2053: RNCamera View Doesn't Rotate

t-hughes, 2019-01-14:
# Bug Report
**To Do First**
- [Y] Did you try latest release?
- [Y] Did you try master?
- [Y] Did you look for existing matching issues?

**Related Modules**

RNCamera

**Platforms**

iOS

**Versions**

- Android:
- iOS: 11.4.1
- react-native-camera: 1.8.0
- react-native: 0.55.4
- react: 16.3.2

**Description/Current Behaviour**

The bug that was reported in #1459 is what I am experiencing. The video provided in that ticket is exactly what is happening. "I found that when the device is rotated the view is initially wrong, but if I start to tilt the device down like I was going to lay it on a table it corrects itself."

**Expected Behaviour**

When the device is rotated the view should rotate along with it.

**Steps to Reproduce**

Open up the camera and turn the device to landscape mode. The view does not change.

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

n1ru4l, 2019-01-17:
@SimonErm can you take look into this?

SimonErm, 2019-01-22:
Yes, i will do.

SimonErm, 2019-01-23:
@t-hughes Can you provide the props you are using?

t-hughes, 2019-02-05:
@SimonErm - type, ref, style, and onBarCodeRead

t-hughes, 2019-02-19:
Update on this, I have used both imports as well. So, both Camera and RNCamera and they both give the same result.

closed 2019-06-08 by stale[bot]