react-native-camera/react-native-camera #2057: How to scale the TextBlock coordinates for displaying

Lythenas, 2019-01-17:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

I'm basically trying to build something like [this](https://heartbeat.fritz.ai/building-text-detection-apps-for-ios-and-android-using-react-native-42fe3c7e339) but without using `react-native-text-detector`. In the blog post `react-native-camera` is only used to take the picture. I use it to recognize the text and display the boxes in real time. This does work however the boxes are offset a bit.

I guessing this is because the image displayed is cropped. Can I get the resolution of the image used for text recognition from somewhere? So I can use them to scale the coordinates returned in the TextBlocks correctly.

[image]

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

n1ru4l, 2019-01-18:
@recepkocur Please stop spamming issues.

closed 2019-06-08 by stale[bot]

acoutts, 2020-01-07:
Hi - was there any resolution for this?

CryptonAdmin, 2020-01-27:
Same story. Did you found solution?