react-native-camera/react-native-camera #2058: How to get face orientation feature

ManigandanRaamanathan, 2019-01-18:
# Feature-Request

**Describe the Feature**
I want face orientation detection feature. Such as Y and R value of Google Mobile Vision https://developers.google.com/vision/face-detection-concepts#face_orientation

**Possible Implementations**
It is possible to implement. Google mobile vision already have these feature and the expo also have these feature in their library https://docs.expo.io/versions/latest/sdk/facedetector called "yawAngle". Therefore, can you please provide this option?

**Related Issues**

n1ru4l, 2019-01-18:
We are open to contributions that implement this feature.

ManigandanRaamanathan, 2019-01-18:
@n1ru4l  Yes, Thanks for help. Looking forward the feature

ManigandanRaamanathan, 2019-02-12:
@n1ru4l  may I know the status of this enhancement? is i possible to get output face orientation whether facing left, right, up or down?

n1ru4l, 2019-02-12:
@ManigandanRaamanathan Feel free to sent a pull request to implement this. I do not have time or the requirements to work on this. Please also avoid generating unnecessary notifications. Once there is any progress I am sure someone will notify you in this issues.

closed 2019-06-08 by stale[bot]