react-native-camera/react-native-camera #2059: RNCamera with orientationFix crashes on Samsung S5 (Android)

shaktalles, 2019-01-21:
# Bug Report
**To Do First**
- [ x ] Did you try latest release?
- [ x ] Did you try master?
- [ x ] Did you look for existing matching issues?

**Related Modules**

RNCamera

**Platforms**

Android

**Versions**

- Android: 6.0
- react-native-camera: 1.9.0 & 1.2.0
- react-native: 0.57.7
- react: 16.6.1

**Description/Current Behaviour**

After taking pictures with RNCamera, I found the rotation issue in several android devices.
Using fixOrientation (#493) solved the issue.
Except when using the back camera on a Samsung S5 running Android 6.0.1.
With flashMode.auto, 2~3 seconds after firing async takePicture, the camera flash lights up and soon the app crashes. Then there was the following YellowBox on console:

"The 'captureAudio' property set on RNCamera instance but 'RECORD_AUDIO' permissions not defined in the applications 'AndroidManifest.xml'. If you want to record audio you will have to add '<uses-permission android:name="android.permission.RECORD_AUDIO"/>' to your 'AndroidManifest.xml'. Otherwise you should set the 'captureAudio' property on the component instance to 'false'."

Asking for the permission on the manifest obviously removed this warning.
With flashMode.off, the app crashes immediately on this.camera.takePictureAsync().
In both cases (flash auto or off) the function doesn't get to finish.
(The app crashes before reaching the standard console.log(data.uri) in the sample code.)

**Expected Behaviour**

Take picture without crashing the app.

**Additionals**

n1ru4l, 2019-01-21:
Please provide crash logs.

shaktalles, 2019-01-21:
Here it is.

[bugsnag_error_stacktrace_java.lang.RuntimeException_event_2019.01.21.txt](https://github.com/react-native-community/react-native-camera/files/2779649/bugsnag_error_stacktrace_java.lang.RuntimeException_event_2019.01.21.txt)

n1ru4l, 2019-01-22:
Duplicate of #1914

closed 2019-01-22 by n1ru4l