react-native-camera/react-native-camera #2060: Unable to resolve module handlePermissions after installing v1.9.0 using react-native link

cquillen2003, 2019-01-21:
# Bug Report
**To Do First**
- [x] Did you try latest release? Yes
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

RNCamera

**Platforms**

Android

**Versions**

- Android: 27
- react-native-camera: 1.9.0
- react-native: 0.57.8
- react: 16.6.3

**Description**
Application will not run on Android because of the following error:

"Unable to resolve module `./handlePermissions` from `/node_modules/react-native-camera/src/Camera.js`

**Additionals**
I've tried deleting node_modules and reinstalling but I get the same error.

sibelius, 2019-01-21:
this was removed from the latest version, check master code

cquillen2003, 2019-01-21:
I tried the master branch as well.  Same error.

sibelius, 2019-01-21:
It was removed

twalk4821, 2019-01-22:
clear bundler cache: npm start -- --reset-cache

n1ru4l, 2019-01-22:
@cquillen2003 Please report if @twalk4821 solution works for you.

**yarn**: `yarn react-native start --reset-cache`
**npm (with start script): `npm start -- --reset-cache`
**npm (without start script): `npx react-native start --reset-cache`

I personally added the `--reset-cache` flag to the `start` script in my `package.json`.