react-native-camera/react-native-camera #2061: Toggling torch works only when remote debug is ON (android 7.0)

aiibe, 2019-01-22:
Toggling torch on/off won't work without remote debugging on.
Happens with Android (7.0)
Not tested with iOS yet

``` jsx
//  ...
import {RNCamera} from 'react-native-camera'

class Capture extends Component {
  constructor(props){
    super(props)
    this.toggleTorch = this.toggleTorch.bind(this)
    this.state = {
      torchON: false
    }
  }

  toggleTorch(){
    if(this.camera){
      this.setState({torchON: !this.state.torchON})
    }
  }

  render(){
    return(
      <View style={localStyle.container}>
        <RNCamera
      	  ref={ camera => this.camera = camera }
      	  style={localStyle.preview}
      	  captureAudio={false}
      	  flashMode={this.state.torchON ? RNCamera.Constants.FlashMode.torch: RNCamera.Constants.FlashMode.off}
      	  permissionDialogTitle={'Permission to use camera'}
      	  permissionDialogMessage={'We need your permission to use your camera phone'}
        />
        <View style={localStyle.bottom}>
          <TouchableOpacity onPress={this.toggleTorch} style={localStyle.switch}>
             <Text style={{ fontSize: 14 }}>Torch {this.state.torchON ? "ON" : "OFF"}</Text>
          </TouchableOpacity>
        </View>
      </View>
    )
  }
}

const localStyle = StyleSheet.create({
  container: {
    flex: 1,
    backgroundColor: 'black'
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center',
  },
  bottom:{
    justifyContent: 'center',
  },
  switch: {
    flex: 0,
    backgroundColor: '#fff',
    borderRadius: 5,
    padding: 15,
    paddingHorizontal: 20,
    alignSelf: 'center',
    margin: 20,
  }
})
```

n1ru4l, 2019-01-22:
@aiibe Are you interested in solving this issue?

aiibe, 2019-01-22:
Issue solved.
I feel so stupid to put ``` console.log(this.camera)``` inside the render function.
I forgot to remove/comment it out causing the app to stuck.
My bad my bad...

closed 2019-01-22 by aiibe