react-native-camera/react-native-camera #2063: Could not resolve project :react-native-camera

aiibe, 2019-01-23:
**Platforms**
Android

**Versions**

- Android: 7.0
- react-native-camera: (master)
- react-native: 0.57.8
- react: 16.6.3

**Description/Current Behaviour**

Build Error
```
* What went wrong:
Could not determine the dependencies of task ':app:preDebugBuild'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > Unable to find a matching configuration of project :react-native-camera: None of the consumable configurations have attributes.
```

Every time a new module is added via ``npm install``, react-native-camera directory will be missing\removed from node_modules.
I have to re-install react-native-camera to solve the problem. Found the solution [here](https://stackoverflow.com/questions/51306490/could-not-resolve-project-react-native-camera)

**Steps to Reproduce**
react-native-camera must already be installed and linked first.
Try ``npm install --save redux redux-thunk`` for instance

n1ru4l, 2019-01-23:
What version of `npm` are you using? You can run `npm --version` to display it.

Do you install `react-native-camera` with `npm install --save react-native-camera`?

aiibe, 2019-01-23:
Yes I used ``npm install --save react-native-camera``
``npm`` version 6.6.0

n1ru4l, 2019-01-23:
I might be wrong but I do not think that this issue is related to `react-native-camera`. Have you tried downgrading your `npm` version? I am using `yarn` and have not used `npm` in a while, but `npm install` should actually work fine.

n1ru4l, 2019-01-23:
You should be able to install the package with `npm install --save react-native-camera@git+https://git@github.com/react-native-community/react-native-camera.git`

You should actually never edit the package.json dependencies and devDependencies manually since the changes will not be reflected in your lock files. Furthermore I think you forgot the `.git` extension in your url.

n1ru4l, 2019-01-23:
@aiibe the documentation needs some improvements then 😊Using the master branch is only recommended when you can live with things breaking.

@aiibe Just out of curiosity: Have you any specific reason for installing from master?

minicase, 2019-04-05:
I have the same issue. RN5.9.3

wesleyguirra, 2019-04-05:
Same issue

closed 2019-04-05 by aiibe

MoizYousuf, 2019-04-16:
Same Issue

ehorodyski, 2019-04-19:
Same issue...used the docs to install. I tried installing it through yarn.

Tino-F, 2019-04-23:
https://github.com/react-native-community/react-native-camera/issues/2210#issuecomment-483677300 fixed it for me

fadlykayo, 2019-04-23:
In android/app build.gradle:

```
android {
  ...
   defaultConfig {
     ...
     missingDimensionStrategy 'react-native-camera', 'general' <-- insert this line
   }
}
```

petarprok, 2019-04-23:
@fadlykayo Thank you this solution worked for me. :)

andrewhartwig, 2019-04-26:
Hmm, @fadlykayo this solution did not fix it for me, on a fresh install of react-native `v0.59.5`

andrewhartwig, 2019-04-26:
hmm, bizarrely, moving it to the top of the `defaultConfig` key fixed it, or i dunno, maybe the file wasn't saved, but builds are passing now :/

johnstew, 2019-05-01:
@fadlykayo just for my own knowledge. what exactly is this doing?

also this work for me so thank you!

moustaphadiakhate, 2019-05-07:
insered like this :
i created in project directory after configured all;
./libs/node_modules/react-native-camera
And in my App.js file
./App.js import Camera from './libs/node_modules/react-native-camera';
Camera bla bla bla ..... working  !

superzhy, 2019-05-14:
Same issue , react-native v0.59.5

superzhy, 2019-05-14:
@fadlykayo  Thank you, This method solves my problem

rwoloszyn, 2019-05-19:
@fadlykayo Thanks, same for me.

VyankeshH, 2019-08-28:
@andreaskeller I also could not resolve the issue,Did you solve this??

ilibilibom, 2019-11-20:
> In android/app build.gradle:
> 
> ```
> android {
>   ...
>    defaultConfig {
>      ...
>      missingDimensionStrategy 'react-native-camera', 'general' <-- insert this line
>    }
> }
> ```

@fadlykayo - why is this not in the installation RNcamera guide?