react-native-camera/react-native-camera #2065: How to change

Dror-Bar, 2019-01-23:
# Question

How to change the path where image is saved using takePictureAsync()?
In the docs I only see the option to change path for video.
Is it possible?

n1ru4l, 2019-01-23:
It seems like this is not possible yet. However we could accept a pull request for this. Shouldn't be to hard to implement. Feel free to send a pull request!

Dror-Bar, 2019-01-23:
Thanks for the answer.
I'm not sure how to do this myself. But for the meantime I might try to use the solution suggested [here](https://stackoverflow.com/questions/48151242/how-to-save-a-picture-to-a-specific-folder-from-react-native-camera-roll/50081434#50081434).

Harsh2402, 2019-04-04:
@Dror-Bar @n1ru4l 
Did you get any solution for the same?

Dror-Bar, 2019-04-04:
@Harsh2402 Yes I was able to save images to a specific path using the File System Access API of the library I linked to (react-native-fetch-blob). There might be other libraries that have access to the FS as well but this one worked well for me.

Harsh2402, 2019-04-05:
@Dror-Bar 
Thanks

closed 2019-04-05 by sibelius