react-native-camera/react-native-camera #2067: Add support for Androidx instead of using support library

gla5001, 2019-01-24:
AndroidX will replace the Support Libraries.

>AndroidX is the open-source project that the Android team uses to develop, test, package, version and release libraries within Jetpack.

>AndroidX is a major improvement to the original Android Support Library. Like the Support Library, AndroidX ships separately from the Android OS and provides backwards-compatibility across Android releases. AndroidX fully replaces the Support Library by providing feature parity and new libraries.

See: https://developer.android.com/jetpack/androidx/

gla5001, 2019-01-24:
see PR https://github.com/react-native-community/react-native-camera/pull/2068

closed 2019-06-08 by stale[bot]

doancongbang1991, 2019-06-21:
`npm install --save-dev jetifier (or use yarn, but install it locally in your project, not globally)`
`npx jetify or npx jetify -w=1 (to specify the number of parallel workers)`
`npx react-native run-android`

oagostinho, 2019-07-10:
Hi

I have problem with AndroidX as you can see in the following issue => [(https://github.com/forcedotcom/SalesforceMobileSDK-ReactNative/issues/134)]

Any idea? Any Help?

Thanks!

gla5001, 2019-07-10:
What version are you using? I dont think the androidx support stuff has been released yet since its a major version change. I dont see anything about it in the releases.

sibelius, 2019-07-10:
comment if this works https://github.com/react-native-community/react-native-camera/pull/2306

oagostinho, 2019-07-10:
Hi

The Salesforce Mobile SDK React Native is using the version 0.56.1.

You can check it out here =>
https://github.com/forcedotcom/SalesforceMobileSDK-ReactNative/blob/master/package.json

gla5001, 2019-07-10:
what version of react-native-camera?

oagostinho, 2019-07-10:
@gla5001 The last one.

oagostinho, 2019-07-10:
@sibelius 

I trying to use the following command to install your PR but yarn dont find or undertand the PR

  yarn add react-native-camera@git+https://git@github.com/react-native-community/react-native-camera.git#refs/pull/2306/head

Any idea?

Thanks

sibelius, 2019-07-10:
use the commit hash

oagostinho, 2019-07-10:
Hi @sibelius 

I did it but without any success.

`Stellaxius-Mac:StellaxiusExpensesAppReact Orlando$ yarn add react-native-camera@git+https://git@github.com/react-native-community/react-native-camera.git#482d6b73c9b8faf98f898cd65947951c097d2c78`

``
yarn add v1.16.0
warning ../package.json: No license field
warning package-lock.json found. Your project contains lock files generated by tools other than Yarn. It is advised not to mix package managers in order to avoid resolution inconsistencies caused by unsynchronized lock files. To clear this warning, remove package-lock.json.
[1/4] 🔍  Resolving packages...
error Couldn't find match for "482d6b73c9b8faf98f898cd65947951c097d2c78" in "refs/heads/docs-camera-roll-save,refs/heads/feat/docusaurus,refs/heads/gh-pages,refs/heads/master,refs/tags/v0.10.0,refs/tags/v0.11.0,refs/tags/v0.12.0,refs/tags/v0.13.0,refs/tags/v0.2.0,refs/tags/v0.2.8,refs/tags/v0.3.0,refs/tags/v0.3.1,refs/tags/v0.3.2,refs/tags/v0.3.3,refs/tags/v0.3.4,refs/tags/v0.3.5,refs/tags/v0.3.6,refs/tags/v0.3.7,refs/tags/v0.4.0,refs/tags/v0.4.1,refs/tags/v0.5.0,refs/tags/v0.5.1,refs/tags/v0.6.0,refs/tags/v0.7.0,refs/tags/v0.8.0,refs/tags/v0.9.0,refs/tags/v0.9.4,refs/tags/v0.9.5,refs/tags/v1.0.0,refs/tags/v1.0.1,refs/tags/v1.0.2,refs/tags/v1.0.3,refs/tags/v1.0.3-1,refs/tags/v1.1.0,refs/tags/v1.1.0-2,refs/tags/v1.1.1,refs/tags/v1.1.1-3,refs/tags/v1.1.2,refs/tags/v1.1.2-4,refs/tags/v1.1.3,refs/tags/v1.1.3-5,refs/tags/v1.1.4,refs/tags/v1.1.4-6,refs/tags/v1.1.5,refs/tags/v1.1.5-2,refs/tags/v1.10.0,refs/tags/v1.10.1,refs/tags/v1.10.2,refs/tags/v1.11.0,refs/tags/v1.11.1,refs/tags/v1.11.2,refs/tags/v1.12.0,refs/tags/v1.13.0,refs/tags/v1.13.1,refs/tags/v1.2.0-7,refs/tags/v1.3.0-8,refs/tags/v1.3.1,refs/tags/v1.3.1-9,refs/tags/v1.4.0,refs/tags/v1.4.1,refs/tags/v1.4.2,refs/tags/v1.4.3,refs/tags/v1.5.0,refs/tags/v1.5.1,refs/tags/v1.6.0,refs/tags/v1.6.1,refs/tags/v1.6.2,refs/tags/v1.6.3,refs/tags/v1.6.4,refs/tags/v1.6.5,refs/tags/v1.6.6,refs/tags/v1.7.0,refs/tags/v1.8.0,refs/tags/v1.8.1,refs/tags/v1.9.0,refs/tags/v1.9.1,refs/tags/v1.9.2,refs/tags/v2.0.0,refs/tags/v2.0.1,refs/tags/v2.0.2,refs/tags/v2.1.0,refs/tags/v2.1.1,refs/tags/v2.10.0,refs/tags/v2.10.1,refs/tags/v2.10.2,refs/tags/v2.11.0,refs/tags/v2.11.1,refs/tags/v2.2.0,refs/tags/v2.2.1,refs/tags/v2.2.2,refs/tags/v2.3.0,refs/tags/v2.4.0,refs/tags/v2.5.0,refs/tags/v2.5.1,refs/tags/v2.6.0,refs/tags/v2.7.0,refs/tags/v2.8.0,refs/tags/v2.9.0" for "https://git@github.com/react-native-community/react-native-camera.git".
info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command.
Stellaxius-Mac:StellaxiusExpensesAppReact Orlando$

sibelius, 2019-07-10:
Check npm or yarn docs

oagostinho, 2019-07-11:
Hi @sibelius 

I have to install directly from your fork. 
"react-native-camera": "git+https://github.com/n1ru4l/react-native-camera.git#feat-breaking-android-x-support",

It seems to work. At least, the compilation goes well and the RNCamera shows up in the app.

Now I have to explore the component inside of my Salesforce Mobile.

Really Thanks

P.S: When this fork will merge to master branch of reactive-native-camera?