react-native-camera/react-native-camera #2069: Missing NSMicrophoneUsageDescription key

orcunorcun, 2019-01-25:
When i try to upload my app to app store connect i got a message from them. So your library have to text purpose string in info.plist for NSMicrophoneUsageDescription key. Can you text it and release again? Thank you?

orcunorcun, 2019-01-25:
By the way I am using "react-native-camera": "^1.9.1" version. And I didn't add the NSMicrophoneUsageDescription key in my info.plist file because i don't need to use that function.

n1ru4l, 2019-01-25:
@orcunorcun Could be related to this line: https://github.com/react-native-community/react-native-camera/blob/406641f25919d00879a99f592a5c0fdc4b72907f/ios/RN/RNCameraManager.m#L389

I added this change recently to prevent crashes when `NSMicrophoneUsageDescription` is not defined in your `Info.plist`...

@orcunorcun Could you try to wrap the this code in `#ifdef DEBUG` to check if apple would accept it afterwards?

Maybe we should only do these checks in debug mode.

orcunorcun, 2019-01-25:
Actually am new on react native so I don't know how can I do that.

orcunorcun, 2019-01-25:
It's my App Delegate file

----------------------------------

#import "AppDelegate.h"
#import <Firebase.h>

#import <React/RCTBundleURLProvider.h>
#import <React/RCTRootView.h>

@implementation AppDelegate

- (BOOL)application:(UIApplication *)application didFinishLaunchingWithOptions:(NSDictionary *)launchOptions
{
  NSURL *jsCodeLocation;

  #ifdef DEBUG
    jsCodeLocation = [[RCTBundleURLProvider sharedSettings] jsBundleURLForBundleRoot:@"index" fallbackResource:nil];
  #else
    jsCodeLocation = [[NSBundle mainBundle] URLForResource:@"main" withExtension:@"jsbundle"];
  #endif

  RCTRootView *rootView = [[RCTRootView alloc] initWithBundleURL:jsCodeLocation
                                                      moduleName:@"RoomsApp"
                                               initialProperties:nil
                                                   launchOptions:launchOptions];
  rootView.backgroundColor = [[UIColor alloc] initWithRed:1.0f green:1.0f blue:1.0f alpha:1];

  self.window = [[UIWindow alloc] initWithFrame:[UIScreen mainScreen].bounds];
  UIViewController *rootViewController = [UIViewController new];
  rootViewController.view = rootView;
  self.window.rootViewController = rootViewController;
  [self.window makeKeyAndVisible];
  [FIRApp configure];
  return YES;
}

@end

n1ru4l, 2019-01-25:
edit `node_modules/react-native-camera/ios/RN/RNCameraManager.m` and wrap the referenced lines in `#ifdef DEBUG` and add a `#else` case to resolve with `NO`

orcunorcun, 2019-01-25:
I think it is alrealdy like that.
------------------------------
#ifdef DEBUG
        RCTLogWarn(@"Checking video permissions without having key 'NSCameraUsageDescription' defined in your Info.plist. You will have to add it to your Info.plist file, otherwise RNCamera is not allowed to use the camera. You can learn more about adding permissions here: https://stackoverflow.com/a/38498347/4202031");
#endif
        resolve(@(NO));
    }
}

orcunorcun, 2019-01-25:
Do you mean i should text like that?

------------

RCT_EXPORT_METHOD(checkRecordAudioAuthorizationStatus:(RCTPromiseResolveBlock)resolve
                  reject:(__unused RCTPromiseRejectBlock)reject) {
    if ([[NSBundle mainBundle].infoDictionary objectForKey:@"NSMicrophoneUsageDescription"] != nil) {
        [[AVAudioSession sharedInstance] requestRecordPermission:^(BOOL granted) {
            resolve(@(granted));
        }];
    } else {
        resolve(@(NO));
    }
}

n1ru4l, 2019-01-25:
@orcunorcun change it to this:

```objc

RCT_EXPORT_METHOD(checkRecordAudioAuthorizationStatus:(RCTPromiseResolveBlock)resolve
                  reject:(__unused RCTPromiseRejectBlock)reject) {
#ifdef DEBUG
    if ([[NSBundle mainBundle].infoDictionary objectForKey:@"NSMicrophoneUsageDescription"] != nil) {
        [[AVAudioSession sharedInstance] requestRecordPermission:^(BOOL granted) {
            resolve(@(granted));
        }];
    } else {
        RCTLogWarn(@"Checking audio permissions without having key 'NSMicrophoneUsageDescription' defined in your Info.plist. Audio Recording for your video files is therefore disabled. If you do not need audio on your videos is is recommended to set the 'captureAudio' property on your component instance to 'false', otherwise you will have to add the key 'NSMicrophoneUsageDescription' to your Info.plist. You can learn more about adding permissions here: https://stackoverflow.com/a/38498347/4202031");
    }
#else
    resolve(@(NO));
#endif
}

```

orcunorcun, 2019-01-25:
Ok i am trying it now :)

n1ru4l, 2019-01-25:
@orcunorcun Thank you for testing this 😊

Can you also please also provide your `RNCamera` component code?

orcunorcun, 2019-01-25:
in my code
------------
`<RNCamera
          style={styles.container}
          type={type}
          flashMode={flashMode}
          ref={(c) => { this.cameraRef = c; }}
          permissionDialogTitle={Strings.Permission_to_use_camera}
          permissionDialogMessage={Strings.We_need_your_permission_to_use_your_camera}
          captureAudio={false}
        >
          {this.renderCameraMenu()}
</RNCamera>`

n1ru4l, 2019-01-25:
Okay I just wanted to check if you set the `captureAudio` property! Seems alright, let's wait what apple has to say 😫

orcunorcun, 2019-01-25:
I think problem solved :) they didn't send me an email about that problem this time. And they applied my app for test flight :) Thank you so much! ^^

n1ru4l, 2019-01-25:
@orcunorcun Thank you for investigating on this! I will try to fix this issue soon. For now you could use https://www.npmjs.com/package/patch-package

The following code seems to trigger Apple and therefore should only be used in `DEBUG` mode.

```objc
[[NSBundle mainBundle].infoDictionary objectForKey:@"NSMicrophoneUsageDescription"]
```

n1ru4l, 2019-01-25:
:tada: This issue has been resolved in version 1.9.2 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.9.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #2070 (2019-01-25): fix(ios): only check permission descriptions availability in debug build
Fixes #2069

It seems like Apple is rejecting apps that access a `Info.plist` usage description that is not available. This conflicts with enhanced developer warnings I implemented in #2048.

To fix that I changed the implementation to only check if the usage descriptions are available in `DEBUG` mode. I also added a note developer warnings, that the app might crash in release mode because using a feature that requires a usage description without having one provided will result in a application crash).
commit 3f8b7954 (2019-01-25): fix(ios): only check permission descriptions availability in debug build (#2070), Fixes #2069

It seems like Apple is rejecting apps that access a Info.plist usage description that is not available. This conflicts with enhanced developer warnings I implemented in #2048.

To fix that I changed the implementation to only check if the usage descriptions are available in `DEBUG` mode. I also added a note developer warnings, that the app might crash in release mode because using a feature that requires a usage description without having one provided will result in an application crash).
closed 2019-01-25 by n1ru4l