react-native-camera/react-native-camera #2072: Bad Camera Ref -> no preview, can't access camera class

twalk4821, 2019-01-26:
# Bug Report
**To Do First**
- [ ] Did you try latest release? yes
- [ ] Did you try master? yes
- [ ] Did you look for existing matching issues? yes

**Related Modules**
RNCamera

**Platforms**
Android

**Versions**

- Android: 26-27
- react-native-camera: 1.8.1 and 1.9
- react-native: 0.57.8
- react: 16.6.3

**Description/Current Behaviour**
Preview loads black. When trying to log out the "ref" to camera component, whole app freezes up. Not even null... just hangs up. Can't load preview, can't access camera class. 

**Steps to Reproduce**
Boilerplate camera code: 
```
export default class Cam extends React.Component {
  takePicture = async function() {
    if (this.camera) {
      this.camera.stopPreview()
      console.log('???')
      console.log('camera: ',this.camera)
      const options = { quality: 0.5, base64: true };
      const data = await this.camera.takePictureAsync(options);
      const image = data.uri
   
    }
  };

  render() {
    return (
      <View style={styles.container}>
        <RNCamera
          ref={ref => {
            console.log(ref)
            this.camera = ref;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={RNCamera.Constants.FlashMode.off}
          captureAudio={false}
        />
        <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
          <TouchableOpacity onPress={this.takePicture.bind(this)} style={styles.capture}>
            <Text style={{ fontSize: 14 }}> SNAP </Text>
          </TouchableOpacity>
        </View>
      </View>
    );
  }
}
```

n1ru4l, 2019-01-28:
Logging the camera object can freeze the thread and as a result crash your application, you should avoid it.

twalk4821, 2019-01-29:
Hi - 

Thanks for the suggestion. Just removed all the logging statements. Now I'm back to the original error (which I should have posted, as well):

```
 Error: takePictureAsync: Expected a Camera component
```
Let me know if there's any other information I can provide you. 
Thanks - 
Tyler

n1ru4l, 2019-01-30:
duplicate of #1931

closed 2019-01-30 by n1ru4l