react-native-camera/react-native-camera #2073: Changing Camera Type from back to front result in black screen in specific situation

TheSolly, 2019-01-27:
# Bug Report
**To Do First**
- [ Y] Did you try latest release?
- [ Y] Did you try master?
- [ Y] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
iOS

**Versions**

- Android:
- iOS:
- react-native-camera: any version above 1.3.1 
- react-native: 0.57.4
- react: 16.6.1

**Description/Current Behaviour**
I am using react-native-camera with react-native-video to record a video and then make the user play it. unexpectedly after i record the video with react-native-camera and play it with react-native-video and return back the to the react-native-camera view and my camera type is set to 'back' ( and only 'back') when i change the type to 'front' i get black screen! If the camera type is set to 'front' and i change to 'back' it is working properly...
It work back normally when i remount the component.
This only happens when i use any version above 1.3.1, tested till 1.9.2.

**Expected Behaviour**

Camera should rotate normally from back to front

**Steps to Reproduce**

Here my Component, just removed unnecessary code

	startRecording = async () => {
		this.setState({ recording: true });
		// default to mp4 for android as codec is not set
		const {
			uri,
			codec = "mp4"
		} = await this.state.camera.current.recordAsync();
		// the above await wont resolve till stopRecording is called
		this.setState({
			recording: false,
			processing: true,
			recordingUri: { uri: uri }
		});
		const type = `video/${codec}`;
		console.log(this.state.recordingUri);

		const data = new FormData();
		data.append("video", {
			name: "mobile-video-upload",
			type,
			uri
		});
		console.log(data, this.props);

		this.setState({
			processing: false,
			showRecording: true,
			videoData: data
		});
	};

	stopRecording = () => {
		this.state.camera.current.stopRecording();
	};

	onLoadStart = () => {
		this.setState({ showPlayButton: true });
	};

	onEnd = () => {
		this.setState({ playRecording: false });
		Alert.alert(
			"Do you want to submit this video answer?",
			"click 'Save' to submit this Video or 'Delete' this recording and try another one!",
			[
				{ text: "Save", onPress: this.submitAnswerVideo },
				{
					text: "Delete",
					onPress: () => {
						this.setState({
							showRecording: false,
							recordingUri: null,
							videoData: null
						});
					}
				}
			],
			{ cancelable: false }
		);
	};

	playVideo = () => {
		this.setState({ playRecording: true, showPlayButton: false });
	};

	cameraTypeHandler = () => {
		let rotateCamera;
		const cameraType = this.state.cameraType;
		if (cameraType === "front") {
			rotateCamera = "back";
		}
		if (cameraType === "back") {
			rotateCamera = "front";
		}
		this.setState(prevState => {
			return {
				...prevState,
				cameraType: rotateCamera
			};
		});

		// const { cameraType } = this.state;
		// this.setState({
		// 	cameraType: cameraType === "back" ? "front" : "back"
		// });
	};

	render() {
		let videoPlayer = null;
		if (this.state.showRecording) {
			videoPlayer = (
				<Video
					ref={ref => {
						this.player = ref;
					}}
					source={this.state.recordingUri}
					paused={!this.state.playRecording}
					resizeMode="cover"
					style={styles.backgroundVideo}
					onLoadStart={this.onLoadStart}
					// repeat={this.state.answerVideoUrl ? false : true}
					// repeat
					// `fullscreenOptions={{ enabled: true }}
					// fullscreen={true}`
					// onLoad={this.onLoad}
					// onProgress={this.onProgress}
					// onError={this.onError}
					// progressUpdateInterval={5000}
					// onBuffer={this.onBuffer}
					onEnd={this.onEnd}
				/>
			);
		}

		let recordButton = <VideoRecordButton onPress={this.startRecording} />;
		if (this.state.recording) {
			recordButton = (
				<VideoRecordButton recording onPress={this.stopRecording} />
			);
		}
		if (this.state.processing) {
			recordButton = <VideoRecordButton processing />;
		}

		let AnswerSubmitStatus = null;
		if (this.state.submittingAnswer) {
			AnswerSubmitStatus = (
				<View style={styles.submitting}>
					<Text>Saving 1/1 </Text>
					<ProgressBar progress={this.state.submitPercent} row />
				</View>
			);
		}
		// if (this.state.submittingSuccessful) {
		// 	AnswerSubmitStatus = (
		// 		<View style={styles.submitting}>
		// 			<Text style={{ flex: 1 }}>Submitted Successfully!</Text>
		// 		</View>
		// 	);
		// }

		let playButton = null;
		if (this.state.showPlayButton) {
			playButton = (
				<TouchableOpacity onPress={this.playVideo} style={styles.playButton}>
					<Image source={playButtonImage} />
				</TouchableOpacity>
			);
		}

		return (
			<View style={styles.container}>
				<RNCamera
					ref={this.state.camera}
					captureAudio={true}
					style={styles.preview}
					type={this.state.cameraType}
				/>
				<TouchableOpacity
					onPress={this.cameraTypeHandler}
					style={styles.cameraType}
				>
					<Image source={cameraTypeImage} />
				</TouchableOpacity>
				{recordButton}
				{videoPlayer}
				{playButton}
				{AnswerSubmitStatus}
			</View>
		);
	}
    }

Here is what happenes:

[image]

TheSolly, 2019-01-30:
UPDATE:
setting those two options inside my `recordAsyc` method fixed the problem for now on v1.9.2!
`
  {
    quality: RNCamera.Constants.VideoQuality["720p"],
    mirrorVideo: false
  }
`

jarvisluong, 2019-03-16:
a reliable temporary fix for this is to force remounting the component <RNCamera />. This can be achieved by setting a different `key` to the RNCamera component

Symous, 2019-03-19:
@TheSolly, any solution here firend?

TheSolly, 2019-03-19:
@Symous 

> UPDATE:
setting those two options inside my recordAsyc method fixed the problem for now on v1.9.2!
{ quality: RNCamera.Constants.VideoQuality["720p"], mirrorVideo: false }

Symous, 2019-03-20:
@TheSolly, Thanks friend, I set captatureQuality to 720p when the platform is iOS and it works.

Thomazella, 2019-03-27:
So basically the repro code is good and no fix so far, just workarounds? Looking for issues to contribute

jarvisluong, 2019-03-27:
> So basically the repro code is good and no fix so far, just workarounds? Looking for issues to contribute

yes if you make a fix we would be really appreciated!

furkancelik, 2019-03-28:
> So basically the repro code is good and no fix so far, just workarounds? Looking for issues to contribute

Yes Expect the arrangement.

winadiw, 2019-04-28:
I have to follow @jarvisluong method which is remounting the `key` properties in order to switch from back to front or vice versa.

duongtungls, 2020-05-15:
> a reliable temporary fix for this is to force remounting the component . This can be achieved by setting a different `key` to the RNCamera component

This answer help me fix this issue. Thank you very much @jarvisluong

iDevo, 2020-06-09:
Hi @jarvisluong! So you are basically generating any string and setting the `RNCamera` `key` prop when also switching the camera `type` right? Because it still leads to an black screen in my case. Thanks!

closed 2020-08-15 by stale[bot]

anyatibrian, 2020-10-29:
Changing the camera refs to make it different worked for me

ShakeelAhmad-dev, 2021-02-16:
Black Screen issue is still exist on Android 10.

"react-native-camera": "^3.42.0",
"react-native": "0.63.4",