react-native-camera/react-native-camera #2079: use another variable name for googlePlayServicesVersion

wadehuang36, 2019-01-29:
# Feature-Request

com.google.android.gms:play-services-vision have different version than other com.google.android.gms modules.  

I tried 15.0.2, but it caused problem with latest com.google.firebase modules
I tried 16.0.0, but here is no play-services-vision:16.0.0
and many versions.  I cannot find a version fit with all dependencies. and end up fork the project and changed gradle.build to read another variable like googlePlayServicesVisionVersion

hariDasu, 2019-02-18:
@wadehuang36 would you mind linking to your fork? I have the same issue integrating with react-native-firebase in order to set up push notifications on android

n1ru4l, 2019-02-18:
@ariDasu check out https://github.com/react-native-community/react-native-camera/pull/2112

n1ru4l, 2019-02-20:
:tada: This issue has been resolved in version 1.12.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.12.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #2112 (2019-02-20): [Android] Add support to specify a different google vision version via gradle properties
Closes #2079.

This PR allows users to use a custom version of `play-services-vision` by adding `googlePlayServicesVisionVersion` to their project properties.

If `googlePlayServicesVisionVersion` is not present, it will fallback to `googlePlayServicesVersion`, and if `googlePlayServicesVersion` is not present, it will fallback to `15.0.2`.

**Example**

```
ext {
    compileSdkVersion           = 28
    targetSdkVersion            = 26
    minSdkVersion               = 16
    buildToolsVersion           = [... cut]
closed 2019-02-20 by n1ru4l