react-native-camera/react-native-camera #2083: Android sometimes throws an exception on startPreview

MaartenR, 2019-01-31:
# Bug Report
**To Do First**
- [ ] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?
Been bumping into this issue since release 0.12. Currently on 1.9.2 and it is still occurring. Didn't see changes in master that would make a difference here.

**Related Modules**
RNCamera

**Platforms**
Android

**Versions**
- Android: 8.1.0
- react-native-camera: 1.9.2
- react-native: 0.56.1
- react: 16.5.2

**Description/Current Behaviour**
Sometimes when the camera preview is started an exception is thrown from native code. This results in the preview window remaining black. The exception is logged by Android, but there seems to be no way to catch this error in react. The `onMountError` callback is not called and the `onCameraReady` callback is called signifying that the camera is ready. Implementing `getDerivedStateFromError` or `componentDidCatch` does not help to catch the error.

**Expected Behaviour**
It would be great if the error did not occur at all, but it seems to happen at a very low level so it may not be possible to prevent it from occurring. The desired behaviour would then be to be able to catch the error or register a callback for dealing with it in react. Currently I cannot give any feedback to the user or attempt a rerender.

**Steps to Reproduce**
I am unable to reliably reproduce this error. It sometimes occurs, seemingly randomly.

**Additionals**
The exception that is printed in the logs:
```
01-30 16:43:58.113  2945  9607 I QCamera : <HAL><INFO> start_preview: 699: [KPI Perf]: X ret = -1
01-30 16:43:58.114  2945  9607 I QCameraHalWatchdog: Stopped Watchdog Thread (2071ms)[start_preview]
01-30 16:43:58.114  2945  9607 E MOTCamDev@1.0-impl: getHidlStatus: unknown HAL status code -1
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: Exception in native call
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: java.lang.RuntimeException: startPreview failed
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at android.hardware.Camera.startPreview(Native Method)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at com.google.android.cameraview.Camera1.startCameraPreview(Camera1.java:194)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at com.google.android.cameraview.Camera1.start(Camera1.java:144)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at com.google.android.cameraview.CameraView.start(CameraView.java:285)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at org.reactnative.camera.RNCameraView.onHostResume(RNCameraView.java:482)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at com.facebook.react.bridge.ReactContext$1.run(ReactContext.java:150)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at android.os.Handler.handleCallback(Handler.java:790)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at android.os.Handler.dispatchMessage(Handler.java:99)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at android.os.Looper.loop(Looper.java:164)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at android.app.ActivityThread.main(ActivityThread.java:6626)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at java.lang.reflect.Method.invoke(Native Method)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
01-30 16:43:58.117  5475  5475 E unknown:ReactNative: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:811)
```

pitchourou, 2019-02-28:
got that too , with latest version 1.12.0 , on android 8 and RN: 0.57

sibelius, 2019-02-28:
@pitchourou can you help us debug this one?

andrycodestuffs, 2019-03-01:
Same issue here

### Versions:
- "react": "16.6.0-alpha.8af6728"
- "react-native": "0.57.4"
- "react-native-camera": "^1.4.1"
- "react-native-qrcode-scanner": "^1.1.0"

My implementation:
```
...
// in the render method
<View style={styles.content}>
  <QRCodeScanner
    onRead={this.sendVote}
    cameraStyle={styles.camera}
    markerStyle={styles.marker}
    showMarker />

  <View style={styles.bottomInfo}>
    <Text style={styles.bottomInfoText}>Helper text</Text>
  </View>
</View>
...

...
// later on, in the styles section
export const styles = StyleSheet.create({
  content: {
    flex: 1
  },
  camera: {
    height: '100%',
    width: '100%'
  },
  marker: {
    borderColor: 'rgba(255, 255, 255, 0.5)'
  },
  bottomInfo: {
    position: 'absolute',
    top: 0,
    height: '100%',
    width: '100%',
    alignItems: 'center',
    justifyContent: 'flex-end',
    paddingHorizontal: 32,
    paddingBottom: 54
  },
  bottomInfoText: {
    color: 'white',
    textAlign: 'center'
  }
})
...
```

StackTrace:
```
Fatal Exception: java.lang.RuntimeException: startPreview failed
       at android.hardware.Camera.startPreview(Camera.java)
       at com.google.android.cameraview.Camera1.startCameraPreview(Camera1.java:189)
       at com.google.android.cameraview.Camera1.start(Camera1.java:140)
       at com.google.android.cameraview.CameraView.start(CameraView.java:287)
       at org.reactnative.camera.RNCameraView.onHostResume(RNCameraView.java:478)
       at com.facebook.react.bridge.ReactContext$1.run(ReactContext.java:150)
       at android.os.Handler.handleCallback(Handler.java:794)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29)
       at android.os.Looper.loop(Looper.java:176)
       at android.app.ActivityThread.main(ActivityThread.java:6635)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:547)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:823)
```

andrycodestuffs, 2019-03-01:
Unfortunately, I don't know why it happens since this stack trace is obtained by Firebase Crashlytics service and I honestly don't know how to reproduce the error with my tests

augusto-altman, 2019-03-27:
We are facing the same issue

handsameliu, 2019-03-29:
Android 6 will also appear, but the probability is very low.

d41z, 2019-04-16:
hey @andrycodestuffs I'm also using Firebase with Crashlytics, could you solve this or someone??

I have had this problem with a user using a Motorola G5 Plus Android 8.0.

andrycodestuffs, 2019-04-16:
@d41z Unfortunately, I haven't found any solution yet

RKucinski, 2019-04-30:
Exactly the same here with :

    "react": "16.4.1",
    "react-native": "0.56.0",
    "react-native-camera": "^1.3.1",

Same stacktrace on Crashlytics 
```
android.hardware.Camera.startPreview (Camera.java)
com.google.android.cameraview.Camera1.startCameraPreview (Camera1.java:187)
com.google.android.cameraview.Camera1.start (Camera1.java:138)
com.google.android.cameraview.CameraView.start (CameraView.java:285)
arrow_right
org.reactnative.camera.RNCameraView.onHostResume (RNCameraView.java:458)
com.facebook.react.bridge.ReactContext$1.run (ReactContext.java:150)
android.os.Handler.handleCallback (Handler.java:739)
android.os.Handler.dispatchMessage (Handler.java:95)
com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage (MessageQueueThreadHandler.java:29)
android.os.Looper.loop (Looper.java:145)
android.app.ActivityThread.main (ActivityThread.java:6934)
java.lang.reflect.Method.invoke (Method.java)
java.lang.reflect.Method.invoke (Method.java:372)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:1404)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1199)
```

AO-owen, 2019-05-30:
same issue
```
"react": "^16.2.0",
"react-native": "^0.53.0",
"react-native-camera": "^1.1.1"
```
StackTrace:
```
java.lang.RuntimeException startPreview failed 
    Camera.java:-2 android.hardware.Camera.startPreview
    Camera1.java:181 com.google.android.cameraview.Camera1.startCameraPreview
    Camera1.java:133 com.google.android.cameraview.Camera1.start
    CameraView.java:281 com.google.android.cameraview.CameraView.start
    RNCameraView.java:442 org.reactnative.camera.RNCameraView.onHostResume
    ReactContext.java:152 com.facebook.react.bridge.ReactContext$1.run
    Handler.java:789 android.os.Handler.handleCallback
    Handler.java:98 android.os.Handler.dispatchMessage
    MessageQueueThreadHandler.java:31 com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage
    Looper.java:164 android.os.Looper.loop
    ActivityThread.java:6944 android.app.ActivityThread.main
    Method.java:-2 java.lang.reflect.Method.invoke
    Zygote.java:327 com.android.internal.os.Zygote$MethodAndArgsCaller.run
    ZygoteInit.java:1374 com.android.internal.os.ZygoteInit.main
```

LucianoTrujillo, 2019-07-29:
Any solutions?

sandyalta, 2019-08-29:
any update about it?

nthegedus, 2019-09-06:
Same issue here!

MateusAndrade, 2019-09-09:
@nthegedus you can provide some example code that results in this?

SamiChab, 2019-09-18:
Same issue here!

TrustTheBoy, 2019-09-19:
Same issue here!

witalobenicio, 2019-10-28:
Same issue here. With v2 and using MLKit

witalobenicio, 2019-10-28:
This seems to happen on older devices and android 7 versions only.

JakielaAdam, 2019-11-04:
Seeing this on Android 8 and 9 as well.

witalobenicio, 2019-11-06:
I "solved" it by forking the repo, surrounding `camera.startPreview`with `try catch` and calling `startPreview();` on `catch` until it works or 100 times.
I know this is far away from a fix, but it worked for me, at least in the older devices that I tested.

usmankhan495, 2020-02-24:
I have also encounter same error.I was using **navigate**(React Navigation) for navigating two screen which was using camera component.Solved the issue by replacing **navigate** to **replace**

SamiChab, 2020-02-25:
The issue should be re-opened, it still happens!
`"react-native": "0.61.5", "react-native-camera": "^3.4.0"`

raducostea, 2020-03-12:
This mostly happens on Huawei P20 lite (HWANE), 3840MB RAM, Android 9

theasc, 2020-09-09:
It happens on Honor 10 on my side (android 10), this issue should be reopened

closed 2021-01-13 by stale[bot]

gustavogialim, 2021-02-12:
Happened here too.
"react-native": "0.59.9", "react-native-camera": "~2.9.0"

hisham-blackbuck, 2021-10-07:
I see the project has new maintainers, can this issue be picked up please?