react-native-camera/react-native-camera #2084: library not found for -lRCTCamera

masrc, 2019-01-31:
Hey, guys! So, this week I'm making the RCTCamera to RNCamera migration.
I've followed [this doc](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md) and on Android its working fine, however, on IOS I got this error: 

[image]

Reinstall pod depencencies and running ```react-native link``` again doesn't show any effects. Any idea about to solve it?

**Version**
IOS: 11.3

"react": "16.0.0-alpha.12",
"react-native": "0.48.4",
"react-native-camera": "^1.9.0"

n1ru4l, 2019-02-01:
@masrc Might be possible that `lRCTCamera.a` is still referenced in the `Link Binary With Libraries` section in the `Build Phases` tab on xcode and cannot be located.

masrc, 2019-02-01:
Yep, its solved!! Thank u

closed 2019-02-01 by masrc

pabloimmediate, 2019-08-23:
Thanks, really useful!