react-native-camera/react-native-camera #2085: Missing NSMicrophoneUsageDescription key (#2)

404sand808s, 2019-01-31:
Hey, sorry for opening another issue with this - tried to comment on the other thread but it was locked to me - but I think the issue may not be resolved. My app is still being automatically rejected because it wants me to have NSMicrophoneUsageDescription in info.plist. I also have put captureAudio={false} in my use of the RNCamera component :)

 Link to original issue [here](https://github.com/react-native-community/react-native-camera/issues/2069) I'm still getting the NSMicrophoneUsageDescription issue when I upload my app to the App Store, and I'm on the latest version of react-native-camera (1.10.10).

Just wanted to put this out there in case it's not just me!

n1ru4l, 2019-02-01:
Related: 
- https://github.com/react-native-community/react-native-camera/pull/2048
- https://github.com/react-native-community/react-native-camera/pull/2070

Because of this code Apple Rejects the app if `NSMicrophoneUsageDescription` is not provided. This sucks since the code will never be called if you set the property `captureAudio` to `false` this code will never be called.

```objc
[[AVAudioSession sharedInstance] requestRecordPermission:^(BOOL granted) {
  resolve(@(granted));
}];
```

I already thought about a way to fix it, however this will be a breaking change: 

By default `captureAudio` is `false`.

If you want to record audio you have to set `captureAudio` to `true` and also add `#define REACT_NATIVE_CAMERA_ENABLE_RECORD_AUDIO YES` to your `AppDelegate.h`.

This would involve some more setup code from the library consumers, however it will allow us to retain the meaningful warnings while not being rejected by Apple.

What is your opinion on this?

**Workarounds:**

- Add `NSMicrophoneUsageDescription` to `Info.plist`
-  Change the code according to this comment: https://github.com/react-native-community/react-native-camera/issues/2069#issuecomment-457507508 (E.g. with [patch-package](https://www.npmjs.com/package/patch-package))

adimshev, 2019-02-20:
Same trouble, version - 1.11.1

n1ru4l, 2019-02-20:
@adimshev Can you please test if #2119 resolves this issue?

adimshev, 2019-02-20:
I already added the NSMicrophoneUsageDescription, uploaded it to App Store and switched to another project. So I can't. Sorry.

n1ru4l, 2019-02-20:
Well I do not have a project where I can test this on. Once someone can we can resolve this issue.

adimshev, 2019-02-21:
I need run `react-native link` again?

augustl, 2019-02-21:
Just tested this - we got past the error message and the app was approved into TestFlight :+1:

n1ru4l, 2019-02-22:
@sibelius What is you opinion on this?

With `[[NSBundle mainBundle].infoDictionary objectForKey:@"NSMicrophoneUsageDescription"] == nil` I can only check whether audio recording is even possible.

However in order to check whether the user has allowed audio recording (or not) there is no way past calling `[[AVAudioSession sharedInstance] requestRecordPermission:^(BOOL granted) {}];`. Which will in return cause a rejected app in case you did not define `NSMicrophoneUsageDescription` in your `Info.plist`.

One solution would be to force users to add a define statement to their app delegate as I mentioned [here](https://github.com/react-native-community/react-native-camera/issues/2085#issuecomment-459628212).

However this would definitely be a breaking change.

A non breaking would  be to check whether `NSMicrophoneUsageDescription` is defined in your `Info.plist` during build time. Maybe this might be possible? Any ideas?

rjblopes, 2019-05-08:
Since I'm not using audio recording I was able to solve this issue by replacing all code inside `checkRecordAudioAuthorizationStatus` with a single line `resolve(@(NO));`. 

Obviously this is not a solution for all use cases.

olegwn, 2019-06-25:
Putting `captureAudio={false}` in RNCamera props solved for me

closed 2019-08-31 by stale[bot]