react-native-camera/react-native-camera #2086: RNCamera.Constants.BarCodeType.upce returns undefined

michalkow, 2019-02-01:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
Android
iOS

**Versions**
- Android: 26
- iOS: 12.1
- react-native-camera: 1.10.0
- react-native: 0.57.8
- react: 16.6.3

**Description/Current Behaviour**
RNCamera.Constants.BarCodeType.upce returns undefined

**Expected Behaviour**
RNCamera.Constants.BarCodeType.upce returns 'UPC_E'

**Steps to Reproduce**
console.log(RNCamera.Constants.BarCodeType.upce)

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

SimonErm, 2019-02-15:
Just a typo, should be upc_e instead. I am going to do a PR this weekend.

n1ru4l, 2019-02-15:
:tada: This issue has been resolved in version 1.11.1 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.11.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #2109 (2019-02-15): Fix BarcodeType type #2086
Fixes #2086
closed 2019-02-15 by n1ru4l