react-native-camera/react-native-camera #2088: Undefined is not an object (evaluating CameraManager.Aspect)

tomderudderetna, 2019-02-02:
# Bug Report
**To Do First**
- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Related Modules**

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**
[image]

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

n1ru4l, 2019-02-03:
Please try to link the project properly with `react-native link` (or manually depending on your setup). Please also keep in mind, that this project could also have conflicts when using the expo SDK.

tomderudderetna, 2019-02-07:
Your example don t work, I have no choice to use Expo beacause create-react-app is is not up to date

n1ru4l, 2019-02-07:
What exactly is wrong with the example app?it works fine for me.

Please provide a reproduction repository.

SimonErm, 2019-02-13:
The stacktrace leads to Camera.js which relates to RCTCamera. Am I right? If i am right try RNCamera, because RCTCamera is deprecated.

n1ru4l, 2019-02-15:
Closing this issue since there is no feedback from the author. If anyone still experiences this issue with the latest version of react-native-camera and `RNCamera` he should open a new issue.

closed 2019-02-15 by n1ru4l